df8oe/UHSDR #731: store step size and mode for each band automatically

DL8EBD, 2017-01-09:
not a must have, but it would be a nice feature to store automatically the last step size (e.g. 100Hz, 1kHz, 10kHz...) and the last mode (e.g. USB, LSB, SAM-L....) for each band separate

df8oe, 2017-01-09:
Mode is already stored for each band (it is stored DÒUBLE - for each VFO seperate...)

Step size does not make any sense - I think. It is not specific to band than to my condition (do I want to tune whole band / do I want to fine tune to station). So please explain what advantage there is when I store it band specific - I do not see that...

db4ple, 2017-01-09:
Mode is stored, sideband is not (except USB/LSB which are two modes technically).

closed 2017-01-09 by DL8EBD

DG9BFC, 2017-01-09:
step size makes sense if you use one mode with a bigger step size (example fm or am) while you prefer smaller steps in other modes (ssb or cw)