df8oe/UHSDR #734: Bad encoder function

oz8gw, 2017-01-10:
I have bad encoder function in all my four encoders. A lot of times when I turn an encoder it does not influence the mchf and a few times it even switches the wrong way.
I have written about about this before and you accepted that it was a software issue. I wonder if you have thought about it in the time between.
Vy 73 de OZ8GW, Leif

df8oe, 2017-01-10:
There are no software related encoder issues known since a few months.

73
Andreas

db4ple, 2017-01-10:
Small addition to Andreas' statement : If it would be a simple and pure software issue, all users would have it more less in the same way. Several months ago we changed a lot of code and it made the use of the encoders for most users fairly unproblematic. 

There is still a single place where I see very small issues sometimes and that is the Encoder 3 when used inside the menu. Here sometimes the change goes into the wrong direction but all other encoders work as expected or I don't notice any issues at least.
We have had one encoder issue in the german forum lately, and it was a faulty encoder. But 4 faulty encoders are unlikely, 4 problematic encoders not so. I have had some cheap encoders in a different project and they were basically junk. Lots of bouncing, next to unusable. Replacing them with a proper brand encoder fixed the problem.  That brings me to the question: What encoders do you use? 

73
Danilo

oz8gw, 2017-01-10:
Hi again.
I also thought that when all four encoders had the failure it must be a sw problem.
My version is 0,4 and I use the encoders that followed the kit.
Vy 73 de OZ8GW, Leif.

db4ple, 2017-01-10:
Hello Leif,

I don't own a kit, so I can't judge. But while it may be possible that the kit encoders are not of best quality found on this planet, if these were really bad, I guess we would here much more complaints. 

I would suggst to post in the Yahoo group and ask if others have similar issues to yours and which hardware and firmware they use. That should see if you are the only one with this problem or if we developers are just lucky with our nice and shiny ALPS encoders. If you are able to run a DSO on the encoder signals, this should tell us something.

73
Danilo

DG9BFC, 2017-01-10:
i also have same thing on vfo encoder ... if i turn a bit too fast upwards then vfo jumps downwards

a few versions back that problem went away (like magic) ... but now it is back (or it seems so)

turning downwards i can turn as fast as i can and have no "drop-outs" ... but when tuning upwards and a bit fast the problem occurs

i can not exactly say if it is back now for one or two days ... (or a bit longer) cause i can not test every version (or only for a short time before going to bed) cause i have not much spare time for testing ( caused by qrl ... grmbl)

df8oe, 2017-01-10:
I think you both have ugly encoders. I cannot confirm issue here.

oz8gw, 2017-01-10:
Hi again
Thanks for the comments. I don't know what a is DSO.
Maybe I should buy four new encoders and change them.
Vy 73 de OZ8GW, Leif.

db4ple, 2017-01-10:
Sorry, digital storage oscilloscope. 

73
Danilo

WD8BXS, 2017-01-11:
I am having the same issue here. I am using a Bournes encoder for the VFO. And the ones Chris supplied with the kit.
DF8OE, can you tell us what decoders you are using?  Thanks, Chuck WD8BXS

df8oe, 2017-01-11:
I am using ALPS encoders. Aren't you the one who is experimenting with encoders without detents? Maybe thats the reason? Try using encoders which are in BOM of Chris.

73
Andreas

IN3VXH, 2017-01-12:
Hi guys,
Same issue on a brand new 0.5 kit with Alps encoders. Sometimes rotating the knobs nothing happens other times it jumps on reverse. It looks like after the command there is some sort of delay on SW side and then the interpretation is not correct.
73
Luca

df8oe, 2017-01-12:
Again:
if it is a software issue MANY MANY (all!!) users must have this issue. But it isn't so! Encoders are working smoothly and perfect since nearly half a year.

You can put mcHF to unusability if you switch on DSP NB or DSP NR. That's a known issue of DSP NR and DSP NB functions. If you have enabled it: turn it off...

73 de
Andreas

DG9BFC, 2017-01-12:
vfo encoder problem is there ... WITH NB OFF AND NO DSP USED!! ... i can live with that (i just turn not as fast as i can and all is well) ....
maybe i test with a different encoder ... andreas ... do you have one as a spare?? or can you order one in your next spare parts order??

df8oe, 2017-01-12:
I can order one next order :)

ffurano, 2017-01-15:
Hi,
also I am experiencing the same wonky behavior on the encoders (all of them), working ok 90% of the times, yet sometimes missing one step or going one to the wrong direction. This is more evident when navigating the menu.
I have the stock encoders, coming with the kit, and a very recent firmware.
Said this, I have no idea if it's a sw or hw issue. Of course it's a minor one, maybe this is the reason why there were not many reports here, maybe users were just thinking that someday it would be fixed, or that it was their fault.
Cheers
f

WD8BXS, 2017-01-24:
My Encoder for tuning acts strange also, I can turn on radio and tuning will be very erratic, goes up when you are going down, or jumps around. just turn off an don again it will be OK for hours.

I don't think it is hardware or it would be the same all the time, and power off on should not fix if it is HW problem!!

df8oe, 2017-01-24:
Hmmm difficult.

There are many, many mcHFs running without this behaviour so I don't think it is a software problem...

I think it is a CONFIGURATION PROBLEM. But I do not have any idea what it can be...

Pse can you try testing using a start with default configuration if it is reproduceable?

73's
Andreas

df8oe, 2017-01-24:
BTW:
Do you have enableb DSP NB or DSP NR?

WD8BXS, 2017-01-24:
No, I do not have DSP NB or NR turned on.When it happens I just reboot.

WD8BXS, 2017-01-24:
I am sorry, I have just changed out my encoder, and all seems to be well at this time.

df8oe, 2017-01-25:
It looks like all encoder problems that exist now are related to hardware problems not to firmware problems. Closed.

closed 2017-01-25 by df8oe

f4fhh, 2017-05-15:
Hi all

I also have the infamous encoder problems. Sometimes there is no increment/decrement from one detent to another, sometimes I get two steps at once. On all 4 encoders. I don't understand what is  happenning, Could my 4 encoders come from a failed batch from Alps ? I replaced the TUNE encoder by a Bourns PEC12R with no detents and it seems to work.

On the Alps encoders the rising edge of phase B is more or less precisely on a detent. On some positions the phase B is 0 on some others it is 1. Phase A is always 1 on detents. On the Bourns encoders the detends are between phases edges.

I tried without success a modified basesw/mcHF/Src/tim.c code : changing FALLING edges to RISING edges and adding some filtering. Next I will try to increase the debounce caps from 1nF to 10 or 100nF. And at least I'll replace the encoders !

73 de Nicolas F4FHH

oz8gw, 2017-05-17:
Hi

Thanks for the tip.

When I look for the encoder Bourns PEC12R I find several types. Can you 
tell me which one you have used.

Vy 73 de OZ8GW, Leif.

Den 15-05-2017 kl. 17:40 skrev Nicolas:

f4fhh, 2017-05-17:
Hi Leif

I got the PEC12R-4020F-S0024 model. This one has no detent and has a push button. You can actually use any encoder from the PEC12R-4***F-*024 series.

I am not sure that it solves the encoder problems, but with no detent it is less noticeable. However it seems that the PEC12R Bourns encoders are of a better quality than the EC12 Alps.

Best 73 de Nicolas F4FHH

poporange630, 2017-11-19:
I can confirm that not only my digitizer plays up this way, the BAND down and up button suffers from the same symptom! When I press down, say from 21mhz, it will jump to18mhz, 14mhz, 10mhz, 14 mhz, 10 mhz and then 7mhz eventually. Please take another look as I kinda believe it's a software issue