df8oe/UHSDR #736: clicking sound when changing volume

DG9BFC, 2017-01-10:
title says all needed
when i turn volume up or down ... every step makes a "click" sound
(no dsp function used!!)

df8oe, 2017-01-10:
I cannot confirm tested with all mcHFs I have.

WD8BXS, 2017-01-11:
When I I get that sound the VFO jumps frequencies radically, but it happens randomly.

df8oe, 2017-01-11:
That's due to bouncing of encoders I think...

If frequency jumps are too big audio is muted during PLL locking time - so you hear clicking when frequency jumps.

WD8BXS, 2017-01-11:
Curious about the number in the subject line (#736) is that the line in the program where this function is?

From: DF8OE 
Sent: Wednesday, January 11, 2017 10:04 AM
To: df8oe/mchf-github 
Cc: WD8BXS ; Comment 
Subject: Re: [df8oe/mchf-github] clicking sound when changing volume (#736)

That's due to bouncing of encoders I think...

If frequency jumps are too big audio is muted during PLL locking time - so you hear clicking when frequency jumps.

WD8BXS, 2017-01-11:
Should the full reset of ser EEPROM delete all of my settings?

I did it, and it rebooted, but all of my settings are still there.

Chuck, WD8BXS

From: DF8OE 
Sent: Wednesday, January 11, 2017 10:04 AM
To: df8oe/mchf-github 
Cc: WD8BXS ; Comment 
Subject: Re: [df8oe/mchf-github] clicking sound when changing volume (#736)

That's due to bouncing of encoders I think...

If frequency jumps are too big audio is muted during PLL locking time - so you hear clicking when frequency jumps.

df8oe, 2017-01-11:
The number is generated from GitHub :)

When you reset serial EEPROM, everything that is stored in EEPROM is flushed. At next start settings which were previously stored in virtual EEPROM are copied to serial EEPROM.

I no EEPROM is detected all settings are stored in virtual EEPROM. It seems that your settings which are stored in virtual EEPROM are bad so you will stuck on them until you do "master reset (load default settings)". This is explained in https://github.com/df8oe/mchf-github/wiki/Operating-Manual

When you have reset all settings you first must power off mcHF. At this moment the default settings are stored. Please notice if the message is very short and tells "...saving to serial EEPROM" or if it tells "saving" with a longer countdown.

If settings are stored in EEPROM you should *not* reset EEPROM because of if you do so your brnd-new settings are overwritten with garbage stored in virtual EEPROM. When storing was done in srial EEPROM the first thing you shpuld do is backing up serial EEPROM (standard menu). So you get rid off damaged settings in virtual EEPROM.

73
Andreas

WD8BXS, 2017-01-11:
By Master reset do you mean I have to go to all the menu settings and press the default button?/

I read the manual and that is all I see.

Chuck

From: DF8OE 
Sent: Wednesday, January 11, 2017 10:51 AM
To: df8oe/mchf-github 
Cc: WD8BXS ; Comment 
Subject: Re: [df8oe/mchf-github] clicking sound when changing volume (#736)

The number is generated from GitHub :)

When you reset serial EEPROM, everything that is stored in EEPROM is flushed. At next start settings which were previously stored in virtual EEPROM are copied to serial EEPROM.

I no EEPROM is detected all settings are stored in virtual EEPROM. It seems that your settings which are stored in virtual EEPROM are bad so you will stuck on them until you do "master reset (load default settings)". This is explained in https://github.com/df8oe/mchf-github/wiki/Operating-Manual

When you have reset all settings you first must power off mcHF. At this moment the default settings are stored. Please notice if the message is very short and tells "...saving to serial EEPROM" or if it tells "saving" with a longer countdown.

If settings are stored in EEPROM you should not reset EEPROM because of if you do so your brnd-new settings are overwritten with garbage stored in virtual EEPROM. When storing was done in srial EEPROM the first thing you shpuld do is backing up serial EEPROM (standard menu). So you get rid off damaged settings in virtual EEPROM.

73
Andreas

df8oe, 2017-01-11:
Search for button combination F1+F3+F5 in operating manual in WIKI.

WD8BXS, 2017-01-11:
THANK YOU, THANK YOU, THANK YOU!!!

That cleared up everything, very nice!

Have a Blessed day!!

73, God Bless,

Chuck WD8BXS

From: DF8OE 
Sent: Wednesday, January 11, 2017 11:10 AM
To: df8oe/mchf-github 
Cc: WD8BXS ; Comment 
Subject: Re: [df8oe/mchf-github] clicking sound when changing volume (#736)

Search for button combination F1+F3+F5 in operating manual in WIKI.

df8oe, 2017-01-11:
Good luck, Chuck :)

starting with a non functional EEPROM *AND* having damaged settings in virtual EEPROM is a very rare combination... I hope you can fix it now!

73
Andreas

DG9BFC, 2017-01-12:
hmmm ... clicking sound when changing volume is gone?!? ... or it is not as strong as it was ....
where there any changes made??

closed 2017-01-12 by DG9BFC

df8oe, 2017-01-12:
Nothing changed here.

WD8BXS, 2017-01-12:
My clicking volume has gone away with the new reset!

WD8BXS, 2017-01-12:
Andreas, when I get my new EEPROM installed, what is the proper procedure ? Should I do a reset EEPROM first before anything else??

df8oe, 2017-01-12:
1. Solder EEPROM
2. switch on mcHF
3. switch mcHF off using power button (don't disconnect power)
4. switch mcHF on again

From now on EEPROM is recognized and used. Configuration is stored in EEPROM from now on (see message at powering down screen). You can do backup from running configuration to virtual EEPROM (using standard menu) and restore it if neccessary (also from standard menu).

RESET EEPROM is only needed if fitted EEPROM was not factory new or used or power connection was lost during power down process.

73
Andreas