df8oe/UHSDR #737: Enhancement

WD8BXS, 2017-01-11:
I think it would be nice to use the snap button to snap a CW signal to match your ST tone. This is the same function that the Auto Tune button on the IC-7600.

Thanks,
 Chuck WD8BXS

db4ple, 2017-01-25:
Please rename the enhancement to indicate the enhancement asked for...

WD8BXS, 2017-04-01:
I guess this should be named auto tune for CW, so when you hit the button it would automaitcally tune the signal to your peak CW tune frequency.

roelandjansen, 2017-05-18:
remove the "guess" part -- it indeed is autotune. It also could be aplied to AM. Just like the Icom TRX we know.

df8oe, 2017-07-29:
I have tried - does not work as expected. This only works if there is no sronger signal in the neigbourhood of the signal you want to snap on. Trying on band with normal CW activities it snaps mostly to the "wrong" signal so it is much better to do this manually.

closed 2017-07-29 by df8oe