df8oe/UHSDR #738: Transverter connection

IN3VXH, 2017-01-12:
Hi all,
I wish to connect 144MHz TRV to low level TX signal and RX input after the PBF.
It will be nice to disconnect the PA when XVTR Offs/Mult is not OFF just to avoid mosfet BBQ.
Can you enter that improvement in the software ?
73 Luca

df8oe, 2017-01-12:
Hi Luca,

it is not possible to disconnect PA via software because there is no technical device like a switch or a relay which can do that.

73
Andreas

IN3VXH, 2017-01-12:
Hi Andreas,
Tnx for fast reply.
Setting Bias to zero ?
Luca

df8oe, 2017-01-12:
This causes decrease of wanted output singnal and increase of significantly distorted and containing harmonics signal. Its level of course is very weak but it consists of mostly unwanted signals - not recommended...

Andreas

DG9BFC, 2017-01-12:
grab signal before pa stage (doing a transverter mod on your own) ... disconnect rfc8 (disabling pa)
it would be possible to mod the mchf that way ... but maybe it is easier to just add an attenuator to your transverter ... and let the mchf run with maybe 2-5w output... that would let the mchf work with a clean (more or less) output signal

closed 2017-01-12 by df8oe