df8oe/UHSDR #739: Recommended MODS question

WD8BXS, 2017-01-12:
Are the recommended MODS for the version 5 board?
Weren't some of them included in the new version?

The mod page only mentions Ver 3 or 4

df8oe, 2017-01-12:
Some of them were included, most of them not. And sadly one mod that is an absolute "no-go" is included in 0.5: Lifetime debugging header is removed and GPIOs are used for touchscreen (which does neither work nor will it be supported in future). Touchscreen GPIOs are used for (not yet supported) WLAN-device :/ So no developer wants to use 0.5 boards because it is nearly impossible to develop firmware without lifetime debugging possibility (and it is not a good ides to diable touchscreen when you want to do debugging: you cannot debug touchscreen with this solution).

So we do not know 100% which mods are included and which not. It is a good idea to comare mod description with actual 0.5 schematic.

73 de
Andreas

s53dz, 2017-01-12:
Well,

Out of the correctness to the author of the mcHF project Chris M0NKA it should here on the Andreas's page be pointed to his document which explains the differences:
http://www.m0nka.co.uk/wp-content/uploads/2016/03/changes_0_5.pdf
I suppose one should first read also other documents before posting biased answers. 
And here, of course, I am not talking about the mentioned changes of GPIOs, inhibiting the debug. Which I do not support.

73 Bojan S53DZ

WD8BXS, 2017-01-12:
Thanks Bojan,
Yes it has been difficult not knowing as a newbie.

I have to wonder if I should stay with 1.4.0.stable version to be safe.

DG9BFC, 2017-01-12:
mod on hardware and firmare changes are 2 different kind of shoes
you can (and should) use newest fw
and you should check what hw mods should be done

WD8BXS, 2017-01-13:
OK, that sound good!!
I guess as long as my radio is working good, I don't really need any more mods hihi

df8oe, 2017-01-13:
It is working - more or less. But maybe you do have distorted TX signal, spurious signals, too low output power and many more "small" issues. There is not ONE reason for all of this and maybe you do not have all of these issues.

Congratulations that your mcHF is working already! Now you have to look at the function in a critical way and look if some of the issues described in modifications is present at your own mcHF. It's time to put it on rail WITH ALL REELS :)

73
Andreas

df8oe, 2017-01-14:
updating modifications page regarding 0.5 boards in WIKI is on top of 2do-list for the next time.

closed 2017-01-14 by df8oe