df8oe/UHSDR #740: line level configurable in menue

DG9BFC, 2017-01-12:
i am using the line level sometimes ... but i have a bit of a problem there (line level is a bit too high)
could we add a menue entry point where the line OUT can be configured (set a bit lower as it is now)
i only see line in setting but it would be nice if i could use line out (and lower the level a bit)
to overcome this problem i now use speaker out (and turn volume down) ... but that is not the solution i am looking for ... so please ... can we add such a menue entry?!? we have line in gain, mic in gain ... but the line out is fixed somewhere in software

df8oe, 2017-01-13:
Line out of WM8731 does not have the possibility to adjust level by software - only headphones out (which is used by LM386 AF PA). So place an attenuator to it if needed.

db4ple, 2017-01-13:
Hi Andreas,
must correct you. Lineout is actually one of the headphone channels (RHPOUT). The WM8371 lineout is used for the TX IQ signal. That would make that possible. Due to the limitations of available on screen space and buttons this will go into the menus.

df8oe, 2017-01-13:
Hi Danilo,

you are right. I was confused bcause of discussions for the "new I40 ui PCB" and said it "by heart" :)

DG9BFC, 2017-01-13:
i know that codec chips headphone out is used ... one channel towards lm386 (and speaker) and the other towards line out ... so i know it can be made configurable by software ... so if it is not a big problem ... please add a menue entry point ... say it is fixed by default (with a given value like 31 step what is max on speaker side) ... but user can change that to spk,31, 30, 29,.....3,2,1 (means one step up it is running in parallel to the volume control!!) ...
then user can turn that menue entry one step up to spk ... and has control over both channels ... or can turn the menue entry down (from 31 default value to whatever he needs as input to his soundcard)
setting default would set the value back to 31 (same level as it is now)
so danilo and andreas ... maybe you can add such a function??

DG9BFC, 2017-01-30:
many thanks for the variable line out level
in some cases it was"slightly overpowered" (distortion in peaks) ... and so i was asking for that variable level ... now we can set it to whatever an external soundcard (or other app) needs
thank you again :-)

WD8BXS, 2017-01-31:
Thank you for the new gain control!!!!
De Wd8BXS
Chuck

closed 2017-02-03 by DG9BFC