df8oe/UHSDR #741: Big Trouble with output

WD8BXS, 2017-01-13:
I got an email from a fellow who said he heard me on 40 and I was 30+ KC wide.
I hooked up with a friend 5 miles away, and here is a picture of my signal on 40 meters and 20 meters.

20 Meters
[image]
All of these lobes are coming from the mcHF

And now 40 the down arrows show the width, and the other arrows show the two lobes of equal strength that the mcHF is puttin out on 40.
[image]

We also checked other bands with similar results.

I have not done the T7 mod yet, will that help with this spurious emission??

Chuck

df8oe, 2017-01-13:
Your output may be overdriven, or PIN diode switching does produce spurious signals. I guess it is PIN diode switching...

Sadly at the moment this modification is only available at our German Project Group website. I will have a look if I can translate it today and put it to the WIKI.

73
Andreas

df8oe, 2017-01-13:
I have translated to modifications page of WIKI:

https://github.com/df8oe/mchf-github/wiki/Recommended-Modifications#ui-0405-h-029

You can test if PIN diode switching is the reason by removing C78 on RF board. If sputious disappear PIN diode switching is the cause. ATTENTION: When removing C78 only TX does work. For testing in QSO you have to use a seperate receiver to listen to your partner ;)

73
Andreas

s53dz, 2017-01-13:
Andreas, 

it should be labeled with RF not UI.

73 Bojan S53DZ

df8oe, 2017-01-13:
You can help us if you correct small typos like this by yourself ;)

DG9BFC, 2017-01-13:
for me the humps every 12 kc looks like not well calibrated iq correction ....

df8oe, 2017-01-13:
...or a disturbed iq signal path (one of the LO signals missing or crippled, one of the four iq audio signals missing or crippled).

WD8BXS, 2017-01-13:
I have tried the IQ correction, but I see no change at all, this could be because of the spurious emission. Maybe after I get the pin diodes fixed the IQ will work better.

WD8BXS, 2017-01-13:
Interesting, the number of humps changes with the RX/TX translation value!!

df8oe, 2017-01-13:
Hi Chuck,

dont do "shots in the dark" - step forward logical...

First measure if both LO mixing signals reach TX mixer with same amplitude. That is a MUST.
Then measure the four audio-IQ-signals which go into TX mixer (use TUNE mode). They must have same amplitude. You can use oscilloscope - and test at a low band (80m or 160m) because of oscilloscope is working there even if its a cheap or old one.

If both is correct, remove C78 and look if distrotions are gone or not. If not (and you do not have idea how to go on) ask again.

Good luck
Andreas

df8oe, 2017-01-13:
I am nearly sure your IQ signal path is broken at some part (LO or audio).

73
Andreas

WD8BXS, 2017-01-13:
OK, are the four audio IQ signal lines labeled TXE_0, TXE_180, TXE_90,TXE_270 ??

Which are the two LO mixing signals??  TX_CLK_P  and TX_CLK_N  ???

df8oe, 2017-01-13:
Correct in both :) On my old 0.4 schematics LO clock is labeled tx_0 and TX_90

WD8BXS, 2017-01-13:
Some Progress my friend, in inspecting the board I found R71 NOT soldered on one side. 
Output looking much better, IQ adjustment still has no effect.
 I will move on to the testing you suggested.

df8oe, 2017-01-13:
IQ adjustment is very sensible and you must follow instructions otherwise you are failing 100%. Please follow instructions 
https://github.com/df8oe/mchf-github/wiki/Adjustment-and-Configuration-Manual#iq-calibration

...before you are looking for an issue what no longer exists...

WD8BXS, 2017-01-13:
Both LO mixing signals equal, all four IQ signals equal.

df8oe, 2017-01-13:
OK - sounds good. Now start calibrating IQ according to link in my last post. If it is working you must be able to suppress the signal which is described in manual down to the noise floor.

WD8BXS, 2017-01-13:
OK, have to go to work, will continue later, Thanks

WD8BXS, 2017-01-13:
Sure would like to see a video of this procedure!

df8oe, 2017-01-13:
have a nice day!

WD8BXS, 2017-01-13:
Hmmmm, I have C78 out but still have receive!!!

df8oe, 2017-01-14:
You see PIN diode switching causes distortions. I think it's the best to fit relay instead of it.
If C78 is removed RX input path is broken. All you hear are very strong signals which are coupling via wires and PCB capacities.

WD8BXS, 2017-01-14:
Yes, I see that.  Where is the proper Schematic for installing the relay?
And I am told that the "Long time in PiplineTX Mod" will fix the extra humps.
http://www.m0nka.co.uk/?p=5545#comment-192171

df8oe, 2017-01-14:
I have added link for getting bigger schematics to WIKI (positioned under embedded small schematics).

df8oe, 2017-01-14:
Because of its not a software related issue I close this now. It is not listed in the active issue list furthermore - but it is still existent and you can write comments and read the existing.

closed 2017-01-14 by df8oe