df8oe/UHSDR #742: curious signs where earlier the band was displayed

HB9LFU, 2017-01-13:
[image]

the last about three firmware releases produce this strange box.

vy73s
Roger

df8oe, 2017-01-13:
What kind of LCD do you use (SPI / parallel, 2.8 / 3.2)?

vy 73
Andreas

HB9LFU, 2017-01-13:
Andreas, I use the small display in parallel mode!

vy73s
Roger

>
> What kind of LCD do you use (SPI / parallel, 2.8 / 3.2)?
>
> vy 73
> Andreas

df8oe, 2017-01-13:
Hi Roger,

I think your configuration was slightly damaged and by setting dBm display to a defined state all was ok...

I don't think we must leave this open (if its not reproducable)...

73 de
Andreas

closed 2017-01-13 by df8oe