df8oe/UHSDR #746: Odd ASCII symbols flashing by

Haasoperator, 2017-01-15:
On the latest daily snapshot (and 1.5.6), ASCII symbols keep flashing by on the blue colored bar next to the bar that shows what output power the radio is set for. WHen I switch from "Scope" to "waterfall", the symbols freeze, but are still there. If I switch back to "Scope", the symbols start flashing by again. Also, when I try to scroll through the bands using the "band+" button, eventually it says "mm" instead of "m" after the band number.

db4ple, 2017-01-15:
Strange symbols: Will be fixed in the next daily
However, the "m" issue should not have appeared in the recent daily builds only in 1.5.6 . Please confirm.

df8oe, 2017-01-15:
One of our Germasn project group does have this, too.

He switsches dmb/Hz - Display in display menu to another item and then back to the one where it was and the issue was gone. No idea what was the cause. Please try this and report.

73
Andreas

db4ple, 2017-01-15:
As I said, it is a bug. This happens if you had new style S meter (dbm or dbmhz) but not enabled the numeric display of Dbm or dbmhz.

db4ple, 2017-01-15:
Please close issue, if it does no longer appear with daily builds.
Thank you!

Haasoperator, 2017-01-15:
Loaded the daily build with yesterday's date. Does not appear to have odd symbols or the "mm" problem.

closed 2017-01-15 by Haasoperator