df8oe/UHSDR #748: Problems LCD

AngelSerra, 2017-01-17:
It's already the third LCD I put and I always have the screen blank.
My configuration is MI2 and MI0 placed resistance sms in as it indicates the lcd plate and the file found in yahoo M0NKA HY28B display.jpg, I asked in yahoo M0NKA that configuration is the correct one for touch screen but I have not found possible solution, they told me that MI2 alone.
How I put them Cutting the printed circuit of the lcd near the pins because neither with desoldador of hot air exits the LCD.

df8oe, 2017-01-17:
Do you want to run LCD in parallel or SPI mode?

AngelSerra, 2017-01-17:
Which would be better of the two configurations?

df8oe, 2017-01-17:
SPI. It is the one which does not waste GPIOs and the first modification which depends on LCD in SPI mode is already out: "acivating real time clock of STM32" :)

Jumpers on LCD are IM0 and IM1 set. You have to do modification for running LCD in SPI mode.

DG9BFC, 2017-01-19:
and here we see it again why not to solder the lcd :-)

df8oe, 2017-01-25:
Good luck! It is a *very good idea* NOT to solder LCD but put it on heaaders...

closed 2017-01-25 by df8oe