df8oe/UHSDR #750: too loud STG in CW when set to 1

LuboKuzman, 2017-01-17:
Hi Andreas, 
is it possible to do some changes in loudness of STG regulation? In evening, when is set to 0 alamost nothing hear, but whe is se to 1 is too loud. And I´ll don´t speak about levels 2 and up. I think, that betveen 0 and 1 should be cca 5 - 10 steps.
Thank you.
Lubo

db4ple, 2017-01-18:
When you set it to zero it is essentially nothing you can hear, since the speaker volume is set to zero.

You are the first one to ask for this range extension. Could you find at least a few others demanding this change (i.e. using the Yahoo group). Technically we would be able to extend the range to give more steps in the lower volume region but this would change the meaning of all the current STG settings and that we'll not do without cause. If it is just,  attenuation of your loudspeaker or headphones would solve your problem as well.

Sorry for no immediate action but I guess you understand

73
Danilo

LuboKuzman, 2017-01-18:
Hi Danilo. Thank you for fast reply. Of course my speaker is almost open at back side with milled openings. With phones is stg ok, so i'll take a resistor and atenuate speaker. 73Lubo                                                                                                                                                                                                                                                                                                                                         Odesláno ze smartphonu se systémem BlackBerry 10.                                                                                                                                                                                                                Od: db4pleOdesláno: streda, 18. januára 2017 21:29Komu: df8oe/mchf-githubOdpovědět: df8oe/mchf-githubKopie: LuboKuzman; AuthorPředmět: Re: [df8oe/mchf-github] too loud STG in CW when set to 1 (#750)When you set it to zero it is essentially nothing you can hear, since the speaker volume is set to zero.
You are the first one to ask for this range extension. Could you find at least a few others demanding this change (i.e. using the Yahoo group). Technically we would be able to extend the range to give more steps in the lower volume region but this would change the meaning of all the current STG settings and that we'll not do without cause. If it is just,  attenuation of your loudspeaker or headphones would solve your problem as well.
Sorry for no immediate action but I guess you understand
73
Danilo

df8oe, 2017-01-19:
I cannot confirm what is told here. If I set STG to "1" it is so weak that I sometimes cannot hear it at all. Clicking of keyer is much louder than STG. I can hear reasonable tone above "5". But a small issue does exist: Gain of sidetone does not change accordingly to encoder - it rests at the level it was before you change it. You have to release key and press it again to activate new value. It would be much better if changings are applied simultaneously to encoder turning without releasing key/PTT. But that's all I see.

73
Andreas

DG9BFC, 2017-01-19:
if it is really to loud for him .. a solution would be he adds a small resistor in front of the speaker amp inside the mchf (damping it a bit) ... then crank up volume a bit to have the same sound strength again
stg should now be softer in setting 1

db4ple, 2017-01-19:
@DG9BFC 
that is what he suggested to do (the resistor thing) to align headphones and speaker. 

@df8oe 
Yes, will do. Seems to have been overlooked during refactoring.

db4ple, 2017-01-19:
Hi,
I tried by myself, and with my headphones 1 was also fairly loud. I am now convinced. So I added 5 more levels down. I also checked, even with 5 levels lower volume everyone should still be able to hear the tone (and then adjust for a more convenient volume). 
If you have already soldered the resistor, I apologize. However, wasn't able to verify until now.

DG9BFC, 2017-01-20:
newest daily built has changed stg strength
so issue is solved

db4ple, 2017-01-24:
Please close since fixed from developers point of view or comment what is still missing.

closed 2017-01-25 by df8oe