df8oe/UHSDR #757: Power on

g7jiq, 2017-01-22:
I am running dalily builds,  and have found that the power button will turn off, but dosn't turn back on,  
Power has to be completly removed and reapplied to work.  If then turned off again the cycle is repeated.
Have rolled back to 1.5.6 and power putton operation os ok.

db4ple, 2017-01-22:
Hi,
this is a known issue with daily builds. 
Since the introduction of a real-time-clock (RTC) modification, the existence of the mcHF RTC Mod is being autodetected. We do this by identifying the mcHF to have a backup-battery connected.

Now, for some mcHF, this detection identifies a backup battery even without this mod executed. In this case the second start tries to start the RTC oscillator. Which fails. The software waits for the  (never starting) oscillator. 

Do you have any modifications in the power supply area which could leave the STM32 3.3V line powered even after turning it off? This would explain the behavoir, since the code expects the processor to be normally fully powered down?

73
Danilo

g7jiq, 2017-01-22:
Hi,
Many thanks for your explanation.
I have no mods to the psu section, I have only done the recomended mods to the rx and swr sections.
I wasn't aware of an RTC mod (must have missed it) and would like to add this.
If adding the rtc mod would be the easiest way to cure this, could you possibly direct me to the mod info,

Cheers 
Dave..

df8oe, 2017-01-22:
Please take a look at the bottom of this page:

https://github.com/df8oe/mchf-github/wiki/Recommended-Modifications

73 de
Andreas

df8oe, 2017-01-22:
Is your LCD working in parallel or SPI mode?

db4ple, 2017-01-22:
Next daily build firmware contains a more reliable detection approach which requires the user to start the RTC. If you don't do this, the mcHF will work as if unmodified.
Instructions in the Wiki have been updated accordingly.

PR #759 (2017-01-22): More reliable handling of RTC Mod Enabling
Now the user has to explicitly start the RTC after detection.
This decision is permanently stored so no further action necessary.
For all user not actively starting the RTC the mcHF will be have as if
not modified (including using the original button mapping).
Presence of VBat Backup RAM is shown in System Info menu and is
precondition for showing the RTC Start Menu.

This should fix #757
closed 2017-01-22 by df8oe