df8oe/UHSDR #762: M2 Button Function

WD8BXS, 2017-01-25:
Sometimes when I press M2 and hold it the screen will switch between DSP OFF and PEAK 750 while still staying blue. Other times, when I press and hold M2, it will just stay on Blue DSP OFF.

What am I doing wrong??

df8oe, 2017-01-25:
It sounds like your button does not pull STM32F4 pin reliable to GND. It may be a bouncing button or a contact sometimes missing at button or at STM32 pin leading to button.

Pressing M2 long is "emergency-off" for ALL DSP functions. Because DSP NR or DSP NB can slow down mcHF in special circumstances to non-operable state you can leave crappy DSP functions by pressing and holding M2  ;)

db4ple, 2017-01-25:
Sorry Andreas,

M2 Press and hold does only one thing from the functional point of view: Enable and disable the Noise Blanker. You can check that in the source code. Exactly as being told in https://github.com/df8oe/mchf-github/wiki/Operating-Manual.

DSP (G2) press and hold deactivates DSP so DSP box will say "DSP OFF" or switch to the last used mode.

Both actions also call the update of the DSP related onscreen graphics. It may be the case that we missed to update the screen at some point so you could see something without doing anything. 

NB function is off if the NB box says "OFF" BTW. DSP functions are unrelated to the NB function. M2 short presses are used to switch between different boxes/values Encoder2 can control. DSP short presses cycles through boxes including the DSP box.

I guess we are talking G2, not M2. Check the manual please.

Thank you!
Danilo

df8oe, 2017-01-25:
Hi Danilo,
sorry - a little bit confused :) I mixed up M2 and G2 :)

After testing M2 I cannot see anything that does not work as expected.

73's
Andreas

db4ple, 2017-01-29:
Hi,
have we an agreement that nothing is wrong here -> close. If not -> more details necessary (e.g. a video showing the issue in action).

Thanks,
Danilo

closed 2017-01-31 by df8oe