df8oe/UHSDR #764: Send Sidetone out to the Line out

WD8BXS, 2017-01-29:
I can see that sending the sidetone level to the line output might be nice so that you can see your own signal on Fldigi.
De Chuck WD8BXS

WD8BXS, 2017-01-29:
Also interesting when I have the line out connected to FLdigi, I see 2 signals. The one on the right in the picture is where I click to copy CW.  If I plug the headphone output into the FDLdigi, I see only the one as normal, and the level is much lower, very nice.
[image]
[image]

db4ple, 2017-01-29:
Hi,
what is the issue about? Apparently using the line out/headphone out gives you the signal you're interested in.
Please clarify.

Thank you!
Danilo

WD8BXS, 2017-01-29:
The bottom picture is using the headphone output. You see only one trace, that is where the CW signal is.  In the top picture using the line out ouput, you see 2 traces.. The right trace is where the CW is copied, the left trace is noise. The line out should show the same signal as the headphone out!!
Also you can see how much more gain is from the line out, TOP picture. Very strange to see 2 traces!!!

db4ple, 2017-01-29:
That is indeed interesting but I guess you should rerun the test now. There has been a very recent (yesterday) change which is supposed to remove some spurs from the CW signal. It may also be the case that also the headphone signal contains the spur but due to the much lower signal level you cannot see it.
The headphone signal ist technically the same signal as the lineout (one comes from the left channel, the other from the right channel of the headphone amplifier). One is I, the other Q  which in case of CW is a sinus of 750Hz (the I and Q phases shift by 90 degrees). So I am sure it is just a visual thing. 
And all the rest goes back to #740 . So you may close this issue (after reporting of the results of the daily build test regarding the spur, please). And watch that line level control gets properly implemented in #740. 
BTW, using the Expert/Debug menu you can make the mcHF also sending the generated signal via USB to the PC and into fldigi (first item in menu).

WD8BXS, 2017-01-29:
Just re-ran the test... Discovered that when the CW peak filter is on you see the 2 traces, when CW peak is off you see only 1 trace..  Strange!!

Line out gain is still toooo Hot!!

db4ple, 2017-01-29:
Okay, 
CW Peak Filter causes that? Strange since not in CW TX signal generation path at all (it is an RX Filter after all).
Someone else needs to verify the issue if possible. CW Peak filter cannot be the cause by itself. Not saying that your observation is wrong though. We need to understand the exact conditions for reproduction of issue. Still the name of the issue itself is completely misleading (the sidetone is send to through lineout, that is not the issue here). Please change, I can't do that.

WD8BXS, 2017-01-29:
Well, the receive audio IS what is coming out of the line out jack. So it has nothing to do with TX

db4ple, 2017-01-29:
??? Please clearly describe the problem you are having. 
What working situation?  In the beginning your where talking about seeing your own signal. Own signal == TX  (at least to me).
Simply describe in full detail how to reproduce the issue and the issue itself. Might sound repetitive since you may have all in above but I can't see it. Others might have the same issue.
If is independent issues, consider creating multiple issues in github.

Thanks.

WD8BXS, 2017-01-29:
OK, i have created another thread for side tone issue, This issue is that when using line out to fldigi, I am seeing 2 traces on the screen (see pics). I have just discovered that when I turn the RF gain down to 25, the second trace seems to go away. This leads me to believe that the line out gain is too high, If we could adjust this level, it may make thing s better.

db4ple, 2017-01-29:
We will check that problem:
So far we have:
RX CW Mode, strong signal, you see 2 Signals if DSP Peak Filter is on and centered on that signal, one signal if off. Right?
Which filter you are using (the one in the lower left blue box) or is this independent of the RX main filter?

WD8BXS, 2017-01-30:
Here is a quick video showing what is happening.

https://youtu.be/1gJOTWKGzKE

df8oe, 2017-01-31:
BTW:
Why are you using analog connections if audio TX *and* RX *and* CAT via one USB cable?

73 de
Andreas

DG9BFC, 2017-01-31:
that also made me wonder a bit ... you can use the one and only usb cable and do not need line in and out ... but ok ... maybe he just wanted to check the function (and maybe he found a bug or not) ...
let me do an educated guess:
line out is (was) very high in level (close to overloading with slight distortion in peaks) ...
adding the cw peak filter add some level to the signal (adding higher peaks) ... and now when you receive a cw note with (example) 600hz ... you will see a second signal at 1800 (third harmonic)
we have no clean sinus signals ... (more round rectangles .... grin) and so a slight overmodulation ... and voila you see third harmonic (of the audio signal) ...
maybe he also overmodulates the line in of his pc soundcard a bit??? that also would cause the third harmonic to be seen ... (you will also see other mixing products and harmnics ... but that is the strongest one to be seen first)

i a quite sure that there is no bug ... but the level in line out was a tad bit high (with default setting)

users that wanted to use line in towards pc sure had those problems ... but never reported cause it was just a tiny bit overdriven and not seen by others very often cause you could lower line in (soundcard sensitivity of pc) ... and since we have usb audio the "normal" line in/out is seldom used for digital work ...
but anyway --- i am thinking that this bug is not a real bug ... but just seeing harmonics from a tad bit tooo high level setting... set that down a bit in menue and all should be well ... if not ... maybe the peak filter adds some harmonics????
only programmer can say that ... so you clever programmers up to the front :-) ... does the cw peak filter adds a third harmonic to the signal??

WD8BXS, 2017-01-31:
Thank you all for checking this out for me. The new line out gain adjust has helped.
Keep up the GREAT work!! I wish I understood half as much as you guys do!!!

de WD8BXS
Chuck

WD8BXS, 2017-01-31:
Sri I did not realize that you could use the USB out to have the same effect, I will use that now.

Thanks,
WD8BXS

df8oe, 2017-01-31:
If you use line out your already digitally present audio signal is converted to anlog in mcHF, then again digitalized in you pc with soundcard. It is much better to use digital signal without double converting :)

73 de
Andreas

WD8BXS, 2017-01-31:
Yes, very true, thank you for the quality information!!

Have a Blessed day!

Chuck de WD8BXS

WD8BXS, 2017-01-31:
This is nice for PSK and RTTY, but if you want to copy CW and use your radio to send CW with your key, you still can NOT see your own CW sending on the screen, I like to see what I am sending,

df8oe, 2017-01-31:
Debug/Exper. Settings --> TX Audio via USB --> Final IQ

This is not a common wish so you must activate it at every start.

73 de
Andreas

df8oe, 2017-01-31:
...and I am not sure if that is working. fldigi may get confused due to RX signals and audio via TX and not synchronizing...

DG9BFC, 2017-01-31:
i do nt know if fldigi does that ... but in multipsk you can use the 
soft for decode ... and connect your key to the pc (sending also via 
soft used as a keyer) ... that w2ay you see both ... received and 
transmitted text (plus you may send some predefined texts from the soft 
like you do in psk also) ...

using usb soundcard is also better because you have no more problems 
with hum noise ... ground loops and such ...

the audio is converted inside the rig so no hum noise can be caused by a 
ground connection towards pc (the pc soft gets digital information so no 
hum noise exists on that data stream)

next is ... fewer chance of overdriving (receive or transmit side

on receive you get the digital data towards digi mode soft (be it 
fldigi, multipsk, mmsstv, whatever) ...

on analog and using headphode towards line in (or mic in if only that 
available) you can overdrive the soundcard and see signals that are not 
there (mixing products, harmonics and such)

on tx (and using analog) your line out to rig may overdrive its mic in 
(with producing overdriven signals as wide as a barn door)

both cases are solved with using the usb soundcard inside the rig ...

receiving end is giving good sensitivity without overdriving soundcard 
levels ... and transmitting end can not overdrive the mic input ... 
cause it is also solved in software (with menue settings and carefully 
setting tx level in  digi modes softwarethere is fewr chance in 
overdriving with wide signal as result)

so ... many advantages in using usb soundcard

....

you also can use the raw iq data and feed that in hdsdr (or multipsk!!!) 
and use the mchf ans sdr frontend (for rx and tx!!)

(hdsdr then acts like a super big screen for the mchf with same viewing 
bandwitdth but better reslution)

or ... you can route the iq data to software "dream" or "sodira" and 
receive drm (digital radio mondiale) ... noise free radio over short- 
and mediumwave (sometimes in stereo!!) .... that is super audio quality 
over a low quality rf path :-)

in dream the frequency control is also done via cat (ft 817 clone) .... 
via hamlib

while in hdsdr sync of sdr soft and rig is done via omnirig driver (also 
with selecting the ft817 as rig)

.....

and many more things to play around with if using the usb soundcard :-)

greetz

sigi dg9bfc

DG9BFC, 2017-01-31:
multipsk des t

hat ... and normally fldigi also should do that ...
you just have to set that in the rig (tx audio towards line out) ...

then fl digi does not know if the audio comes from receiver ... or from 
your key (and side tone generator)

dg9bfc sigi

ps i do not exactly know if the stg also send audio over usb .... but we 
soon will get anwer about that :-)

df8oe, 2017-01-31:
In CW TX signal is generated by sidetone modulated to non-translated LO. That is the only way to get I and Q audio signals out of stereo codec (only two channels!) *and* getting a sidetone simultaneously. So it is tricky CW generation :)

db4ple, 2017-01-31:
Indeed. Tricky business CW generation that is.
However, via USB we have the full freedom to sendback the "TX audio" anytime. 
Right as explained above we use it for debugging the signal. We could think of providing "TX Lineout" via USB. That makes sense for all modes with input NOT coming from USB itself, would allow you to capture whole QSO with the mcHF USB as monitor output.

73
Danilo

DG9BFC, 2017-02-03:
so if you then let a recording soft run (normal windoze wave recorder or similar) you should have rx and tx signal (complete qso) on the recording (be it in ssb or in cw!!) .... yes that would be nice for play back of that audio later (show a friend that you really worked that rare dx location with just qrp rig)
is that possible for ssb also (freedom what to send via usb cable) ???
then yes please add that function (would means send sidetone over usb when in cw or send mic audio when in ssb mode)
it would then also be possible to resend audio (show the other station how you received him) with playing back that recorded sound in the pc (while pressing ptt mchf should "listen" on the usb cable if there is some tx signal and use that) ... just record the rx audio .... replay and press ptt (but change input to usb audio before resend ... voila you send the recorded audio ... so ... if i am thinking right ... that already is implemented in some way .... but only for rx audio ... now add the side tone generator towards usb cable ... and/or  add also the mic audio towards usb cable (so then you can record complete qso)

WD8BXS, 2017-02-04:
I am using the  Debug/Exper. Settings --> TX Audio via USB --> Final IQ 
it will not save that setting, I have to set it every time I turn on the radio!

Any ideas?

db4ple, 2017-02-04:
That is expected, generally debug settings are not persistent, i.e. a reboot will reset them. 
For this particular case, we might in future added the transmit ouput control in the standard menu. As mentioned abobe.

df8oe, 2017-02-09:
If it is ok now please close this issue.

closed 2017-02-09 by WD8BXS