df8oe/UHSDR #766: RIT/MIC

hb9trt, 2017-01-29:
Display Error. If L3 is beeing pressed, it changes between RIT or Mic gain. But the color of RIT don't changes and so you don't know if it's set to RIT or MIC... 

(Wenn die Taste L3 gedrückt wird, dann ändert zwar die Farbe von MIC auf gelb, aber RIT bleibt ebenfalls gelb. Wenn man die Taste nochmals drückt, dann sind RIT wird MIC nciht mehr grau. D.h. beide bleiben gleich und es sieht aus als wären beide Funktionen aktiv).

hb9trt, 2017-01-29:
Tested at the version of january, 25th.

hb9trt, 2017-01-29:
Thank you for the fast fix!

db4ple, 2017-01-29:
Well, was a simple one to fix. And defect was introduced by me a few days ago (only to show how fast we can fix things ... )

closed 2017-01-30 by hb9trt