df8oe/UHSDR #768: Side tone to Line out?

WD8BXS, 2017-01-29:
Can side tone be routed to the line out?

If I use headphone jack out to the fldig I see the side tone, but not out of the line out jack. only receive signal.
Also need a level adjust for the line out.

db4ple, 2017-01-29:
https://github.com/df8oe/mchf-github/blob/active-devel/mchf-eclipse/support/ui/menu/ui_menu_structure_mdtable.md    search for "TX Mute LineOut"

WD8BXS, 2017-01-29:
I am not talking about transmit levels!! Just receive levels.  Issue is not closed.
 What don't you understand?

db4ple, 2017-01-29:
Sorry about not picking up that is was connected to receive.  Lineout always get the same audio signal just as the speaker / headphone gets during RX.
Difference is only that it has a fixed level. 
I am still not clear what you are talking about, so lets define what I mean by TX (transmit) and RX (receive)

RX: you are listening to the received signal from a remote station via antenna (red led is off)
TX: you are transmitting a signal (the red led is on). 
- If you hear something through the speaker but don't get the same signal on lineout (at lineout level) this is a bug. We will take care of that.
- If it is something else, please explain it so that I understand it or you will have to wait for someone else to continue with you since in this case I stop here since there is no point to continue with me if I don't get the point.

WD8BXS, 2017-01-29:
OK,  RX red light off. If I plug into headphone jack, on fldigi I see the received CW signal 
TX red light on, I see my side tone on fldigi when I send CW. What I am sending prints out on screen

If I plug into the line out jack on RX red light off, I see CW signal on fldigi.
When I TX red light on, I do NOT see my CW that I am sending. The side tone does not go to the line out jack when sending TX CW

db4ple, 2017-01-29:
Here we are again: that is controlled by TX Mute LineOut. Setting this to Off will give you the signal on lineout during TX which is what you want. 

However, if you try that now, you will find out, it does not work as advertised. Since there is indeed an issue here after looking at the code which does it. If we would have tried TX Mute Lineout some hours ago, the fix would already be here... It is extremely important to clearly describe all the steps necessary to reproduce a problem. Can't be stressed often enough...

I will fix that problem and Andreas will merge it in a few hours I guess.

WD8BXS, 2017-01-30:
That sounds great, I will check it out on the new daily build!! Now we can close this thread as well,

 Thanks again.
de Chuck WD8BXS

closed 2017-01-31 by WD8BXS