df8oe/UHSDR #771: Attempts to optimize the Rx gain / noise balance

yo8can, 2017-01-31:
I made few tests in order to find a way to reduce the amount of internal noise of my mcHF, feeling that the receiver seems to have an oversized gain at some stage, this conducting somehow to an improper similarity in behavior with a FM processing Rx, who's internal noise expands strongly when the strength of the signal lowers below some lower limit so there, the IF limiters can not do properly their job (to kill the AM parasitic modulation=noise for FM modes). In the mcHF case, being designed especially for amplitude modulated modes (like SSB etc.), the gain distribution in the Rx path is more critical, being a subject to optimizations in order to get the maximum of sensitivity, at a minimum internal noise, besides maximum IP3.  At my mcHF seems to be a problem with the noise at weak signals and more with antenna disconnected, when I may want to use the Rx also for arbitrary and low precision RF measurements benefiting from the 48 kHz spectrum scope, which can be useful in some cases (to test oscillators etc.), when a better spectrum scope is not available in the shack. When receiving strong signals, do to AGC, the gain is reduced automatically and also the internal generated noise contribution to the received signal becomes insignificant, but at weak signals near uV range the gain raises high enough for the internal noise to appear and cover the signals, upon my opinion, to much. This phenomenon occurs in any type of AGC controlled receiver, but at my mcHF, it is too high, the S-meter never shows les than S3 (or S2 and 1/2), without antenna. On the other hand the S-meter indications at strengths lower than S9 are not accurate, and irrelevant for signal strengths under S3, because mostly the internal noise generated by the OPA chip is revealed (in my case, after a previous made test).  With antenna it is true that this internal noise is left behind, do to the AGC effect but the indications of the S-meter seem to be not accurate (over-sized). Not tested the S-meter yet with dBm scale, but I would like that the meter should work well also in the entire S degrees scale, helpful for giving reports.
For a test, I bypassed the OPA amplifier, connecting directly the outputs of the QSD (the 22nF sampling capacitors) to the I and Q inputs of the UI board, disconnected from the OPA outputs (which remained completely insulated).  The S-meter indications are now S1 plus one dot of the bargraph (lighting intermittently), as it should be in a good gain balanced receiver, without antenna connected. When connecting the antenna, the indications of the S-meter shows S6...S7 (on lower bands) in absence of signals, do to band noise. When tuned to a signal the receiver sounds more clean as before (when using the approx. 23dB gain of the OPA), now lesser noise, approaching to a normal and desired behavior. To test the S-meter accuracy, I disconnected the antenna and use some test rig (a variable signal generator in metal shielded case with internal step attenuator, a millivoltmeter to confirm the output attenuated signal delivered by the generator, followed by a -40dB fixed external metallic shielded RF attenuator, all cabinets grounded, using for signal transfer 50 ohm coaxial cables with BNC connectors, in order to minimize errors by parasitic radiation), and find out that the receiver in lower bands without the OPA amplifier still meets the required sensitivity according to the S-meter indications: S4 indication for 1.3uV injected, S5 for 2.6 uV, ... S9 for 50uV at the antenna socket, but for S9+20dB the signal injected was 900uV instead of 500 as it should, but the most important thing is that the indications are good in the S4...S9 + span. Tests where made at 7.1 MHz and bandwidth 2.3 kHz, as usual for normal SSB receiving in comfortable conditions. Between 3.5 and 10.1 MHz the results are in the same range. Not tested over 10 MHz. The software is 0.219.27.0, translate mode ON. I will upgrade next days to the actual stable bootloader and firmware.
According to this simple tests, results:  
1. The maximal gain of the Rx appears over-sized, especially for lower bands; the most elegant method to control this may be in software, by implementing a sort of upper limit of gain setting that the AGC should raise, with zero or 1 S-meter indication without antenna, separate for each band (similar to the existing Tx power setting, but here can be enough only 3 menus, for bands up to 7 MHz. another for bands between 10 and 18 MHz, and the third for bands over 21 MHz), in order to optimise the noise performance of the Rx and S-meter for each band group or local condition; unfortunately I have no knowledge in software programming; please note that the existing RF gain control for this purpose is not suitable, because of the effect to the S-meter indications: lowering the RF gain, the indication of the S-meter raises and cannot give an accurate indication of the strength of a weak desired signal, it conducts that the S-meter should indicate always S8..S9+... or so, for all weak signals of the band, which my be not real, considering also the level tests above mentioned.
2.  For the QSD preamplifier, is better to use a chip with lowest internal noise, like LT6231 which costs less than the OPA2350, I will test this chip after purchasing one;
3. The gain of the low noise QSD preamplifier should be reconsidered, to an optimal value in order to obtain a good balance between sensitivity and noise, a simple scheme with jumper switched precision resistors may be suitable, to adjust the optimal gain (if no software adjustment will be made for maximal ACG gain setting for each band group);
4. In order to maxmize the IP3 it is a good idea to lower the gain of the RF preamplifier at the bands below 18 MHz, like the scheme around Q1411 of the IC7300, or with a simpler circuitry, made around the existing BFR 93 scheme, by interrupting the connection to GND of the C66 capacitor, using a PIN diode switching circuit, which should connect to GND this capacitor only when the 17-15-12-10m LPF is engaged, based on the fact that it is connected to GND, from the point of view of DC (using some RFC RF separating circuitry). So no other port of the MCU is necessary for this purpose.
Regards, 73's, Romi YO8CAN

DD4WH, 2017-01-31:
Hi Romi,
some of your questions and concerns will probably be answered here:
https://github.com/df8oe/mchf-github/wiki/S-Meter-&-dBm-Hz-display

I agree that the S-meter in "old school mode" (--> see menu) is not accurate at all and cannot be used to make any reliable measure of signal strength. But we had to keep that S-meter mode, because there was a demand of several users for an S-meter with such a response. Try the S-meter setting "dBm" which should be as accurate as it can be in an IQ SDR receiver (also set the spectrum magnify mode to 1x --> only then the dBm dispay AND the S-meter is accurate).

IP3 is mainly a function of the preamp, I certainly agree with that and the original preamp is not optimal, thats for sure. But as far as I know, the correct way of getting a higher IP3 would be to eliminate the preamp and NOT to remove the OP-Amp after the QSD. Tayloe has made experiments showing that the QSD itself has an IP3 of >+30dBm. So eliminate your preamp and enjoy the high IP3 ;-).

I do not get your point with the OP-Amp and its noise, that is surely never an issue. Switch off your AGC, and you will never have problems with noise coming from the OP-Amp. By the way, the signal strength that is measured in dBm is dependent on the bandwidth --> more bandwidth, higher signal 
this is of course also reflected in the S-meter. So an assumption of a baseline of "S 0" or something is always wrong, it depends on the antenna noise, band noise, your bandwidth etc. That is normalized, of course, if you use the dBm/Hz display.

Please also always use the newest firmware! Do not use old firmware for creating issues here, thank you!

73 de Frank DD4WH

df8oe, 2017-02-01:
Due to our release policy it is much better to use the "newest daily snapshot" instead of the "last stable version".

Stable versions are daily snapshots which by chance were released as "versions" (testing or stable). There is NO difference in stability or possibility of containing issues - all binaries (daily snapshots, testings and stables) may contain issues (and do contain issues). There is nearly no software on world that is complete free of issues but we are working very intensive that an issue which has high priority is eliminated ASAP. But daily snapshots always do contain all new features and bugfixes. Regarding the speed of firmware development some days can move big rocks so stables are partwise outdated after a few days...

73 de
Andreas

s53dz, 2017-02-01:
Hi Romi, Frank,

I find it a pity that this important information get scattered between different places on forum and git-hub.

For a very good explanation of RF LNA gain and a IF (baseband) LNA gain influence I would suggest you to read a post (if you already have not) from Yves (Chris's M0NKA yahoo group post nr 8883). So, I agree with him on that: there should be a balance in both for the best sensitivity/dynamic range.

Frank,
I like the added signal strength meter, beside conventional S-meter.
BTW: I do not understand why in mcHF as at present state we simply do not get to a common understanding to align S-meter to an accurate dBm reading, since if we decide to comply to a standard for HF S-meter these are THE SAME power levels. (I am the one that made a mistake once when I believed the S-meter is accurate when doing a measurement!)

Additionally I have a proposal to change the term signal strength into a more appropriate (Spectral) Power density (dBm/Hz) as opposed to the Power in dBm. Thus also some questions about the different behavior of both can automatically be omitted.
https://en.wikipedia.org/wiki/Spectral_density

73 Bojan S53DZ

df8oe, 2017-02-01:
Hi Bojan,

structure of a newsgroup (and explicit structure of Yahoo NG) is a very worst place to store such informations. Git Hub is working very well but I never have seen any comment or question of Chris M0NKA and I am not sure he is reading here also. That is not good for the whole project - you are right.

But it is impossible to work together in software devlopment without the benefits of a platform like GitHub or SVN - again: it is *IMPOSSIBLE*. Firmware development never would be at the recent level if it would be maintained as a one-man-show all the time. No one has time to do all that was was done and no one can merge changings in the way Github does this job.

As long as it is not clearly stated from "official site" that firmware related questions and WIKIs do exist here, it is not easy to communicate with scattered postings in Yahoo NG. And I stick up in this point...

73 de
Andreas

s53dz, 2017-02-01:
Andreas,

I can easily go with what you said, but,
you know my standings about the mcHF project already:

mcHF author is Chris M0NKA, the first decent FW provider was Clint KA7OEI, the project manager of a new age mcHF FW/HW modifications is Andreas DF8OE and Co.

I understand that you find the Chris's mcHF HW platform attractive since you work so hard on it, otherwise you would build one of your own and start from there. I suppose I am right saying that.

So please, get you some time and sit down to have a beer or two with Chris. Hi.

It would be easier for us followers to see an aligned activity and progress here!

Writing this in good will,
73 Bojan S53DZ

PS: An open source project certainly has its benefits, however every stick has at least two ends.
Managing that, one has to be wise enough to hold it together and not to burn in the speed.

yo8can, 2017-02-02:
For software developers: in order to optimize the receiver gain / noise balance in the entire HF spectrum, where different maximal gains are suitable according to different band background noises and signal strenghts (the lower bands have more noise and the upper ones are more quiet, needing/allowing  also more Rx gain, please consider the proposal of introducing minimum 3 new  final user settings:
SET MAX. RX GAIN 160-40m, 
SET MAX. RX GAIN 30-20m, (or 30-17m), and 
SET MAX. RX GAIN 17-10m (or 15-10m)
The default settings should be maximal gain to all bands, like is now, but the user will have the possibility to limit those maximal gains according to local condition. It is not good that in a low fequency band in absence of a signal the gain of the receiver to raise at some very high and unusable values, the rezult been nothing else than more noise, against the feeling of a clean, quiet receiver.
It is true that the AGC manage this in a manner, but the result is a feeling of excessive noise in lower HF bands, when propagation conditions are bad or antenna used is small. Please consider that some good industrial made transceivers have separate preamplifiers for low or high frequency bands, and also, different maximal gains.

df8oe, 2017-02-02:
NO - actually not and there was no wish from others before to have such a setting. It would destroy our beautiful dBm display because levels would not match any more. I by myself do own 4 mcHFs and never have had any problem regarding noise. So this is not at the 2do list at this time.

There is already a RFG setting which works fine and you can use manual gain control if you do not like automatic gain control.

I cannot confirm much more technical noise coming from mcHF hardware on lower bands thaan on higher bands. (Of course I am *not* talking about QRM/QRN - but these two are not a reason to change gain on lower bands).

Please be so kind and try first using recent firmware. We only can work on issues of recent firmware.

If you do have issue using recent firmware you can reopen issue report. Before reporting please take care to setup every neccessary configuration (e.g. IQ settings) and verify that your hardware does not have any issue. We only do work on firmware issues.

DD4WH, 2017-02-02:
Hi Bojan,
- I do not have access to the Yahoo group, so could you please copy the explanation by Yves here for better understanding?

 - Regarding "common understanding of S-Meter": an S-Meter is an S-Meter. And as you state, an S-meter has a standard and we stick to that standard (as far as the accuracy of such a very simple design as the mcHF can achieve that). The only confusion that arises is that the "old school S-meter" isn´t really an S-meter at all, because it does not stick to the standard. But due to many users demanding the "old school S-meter" style, we decided to have that option there. So there is no problem with common understanding, just a problem of the user to differentiate between old school and S-meter: details are here: https://github.com/df8oe/mchf-github/wiki/S-Meter-&-dBm-Hz-display

-  "mcHF author is Chris M0NKA, the first decent FW provider was Clint KA7OEI, the project manager of a new age mcHF FW/HW modifications is Andreas DF8OE and Co." --> yes, you are absolutely right: Chris is the first author (and there are many other authors that followed him and are now also authors), and Clint greatly improved the first firmware versions, without KA7OEI, the firmware would never have achieved the state that it has today. These contributions are greatly acknowledged by us in the code (search for Clint/KA7OEI in the source code), in the startup screen of the mcHF and in many of the explanations in the wiki! But well, both the hardware and software was put under the open source licence by Chris in purpose, so take it and modify it! That´s what we do and there is nothing wrong with that, neither from a legal nor from an ethical perspective, I am deeply convinced of that. So: go open source and do not create artificial boundaries, that´s not the purpose of the open source idea, rather the community should improve the design and that´s exactly what Andreas is coordinating ;-). 

- Thanks for the suggestion to rename the dbm/Hz spectrum, will do that: "PSD - power spectral density"

All the best, 73 de Frank

DD4WH, 2017-02-02:
Hi Romi,
cannot confirm the observation of excessive noise when changing bands. Must be an issue of AGC settings:
Maybe you try manual (AGC = OFF) and then use RFG to adjust noise according to your antenna setup/local noise conditions. That should solve your problems.
I am not sure, but maybe you are confused by the name "RF gain" --> as far as I know, that is not hardware gain of some preamp or something, it is 100% software gain. So this gain does not influence IP3 or dynamic range at all.
I am not aware of commercial transceivers that have different preamps, could you name those in detail? 
73 de Frank

s53dz, 2017-02-02:
Frank,

I agree to that big time. As I said, open source and a git-hub, as an excellent tool for that, is good. But, sometimes, if one is too passionate about that, it can somehow limit the others. Just hope that I did not offend somebody saying that. Anyway I can easily cope with all that, however I can also understand those who have difficulty navigating among all the mods and features and places and FW versions.

About S-meter. Yes, the legacy here is strong. The dispersed S-meters results (S-units vs dBm) also, as a research on some link so nicely put. But again, for me the S-meter (S-units for HF in this case!) should be complying the std. And mcHF does offer a vy good base to have it both aligned. If this "old-school" was a demand from some hams, I can not argue, be it. But technically it should have been the same.

Here in the following lines down I try to represent the Yves's comment on gain in an rx chain.

73 Bojan S53DZ

---

8883Re:: Re: : Re: [M0NKA-mcHF] Re:: Attempts to reduce the internal generate
Expand Messages

    roschi_ch
    31 Jan
    Hi Romi,

    very interesting you analysis.

    While I agree that the gain of the baseband LNA could be lowered, I never found the effect realy disturbing.
    My mcHF has a measured MDS of about -130 dBm at 1.4 kHzBW.
    The AGC action is never unpleasent with Max RF gain between 3 and 6. FW Daily snapshot DF8OE.

    Which FW are you using and which MaxGain?

    I reduced the LNA amplification by 6dB (R43 and R44 1k) and the gain of the RF LNA to about 13 dB - with no apperent loss of sensitivy. But the classic S-Meter indication is about 1.5 units too low.
    The reason to reduced the baseband LNA is to prevent clipping of the LNA at high RF levels, I wanted no clipping before -10 dBm. With MaxGain set to 6 (Minimum) there is no loss in sensitify.

    To reduce the RX noise it was important to filter the RF LAN power supply with an extra 10uF ESR capacitor.

    An LNA with lower noise may be beneficial, I'll order an LT6231 and in a few weeks will run some tests.

    A reason why I think the baseband LNA should not have a too low gain: we want to operate the ADC close to their full scale level to obtain best linearity (and to overcome the noise from the digital circuits). Lowering the noise of the baseband LNA should allow to reduce the gain of the RF LNA and thus to improve the dynamic range which actually is limited by the BFR93a LNA.

    carry on, 73, Yves

yo8can, 2017-02-02:
Hi, Frank
I will go on with more tests after firmware upgrading. Regarding industrial transceivers that use separate RF preamplifiers for different band groups, at this moment I can confirm IC7410, IC9100 and TS990s, but are not the only ones.
The service manuals are free to download, IC7410/9100 use for the range 1,8-21MHz a Norton push-pull scheme (lower gain and high IP3, Q511&Q512), and for 24-50 MHz other preamplifier, single stage (Q532) with higher gain. TS990 uses also diffrent preamplifiers, also a pus-hpull Norton scheme with gain of 9 dB for frequency range up to 21 MHz, and a single stage with gain 22dB for 21,5-60MHz range.

DD4WH, 2017-02-02:
Thanks, Bojan, for the info by Yves and thanks Romi, for the info re: different preamps in other transceivers!

It´s nice to see that the mcHF is compared with commercial transceivers costing 5 to 15 times more than the mcHF ;-) 1800,- / 3200,- / 6500,- € [IC7410/IC9100/TS990s].

We can tweak a lot in software for free, but hardware tweaking (pre-amp, better OP-amps, variable gain in the QSD) is often connected to higher cost. But nonetheless we should seek for optimal combination of cost-efficient mods and quality of the TRX.

Have fun with the mcHF, 73 de Frank

DG9BFC, 2017-02-03:
if a very clever rf guru finds the final solution (after doing a massive number of tests and trying different mods) ... that solution is sure just a few dollars in costs (changing a resistor here and there ... using a lower noise opa ... etc) ... rest can be done in software (and i know you all made wonders in the soft!!!)

without using github and community is working together on the soft ... i am sure the mchf would not be 10% as far as it is now

df8oe, 2017-02-04:
In our recent daily snapshot I changed setting range for MAX RX GAIN from 0...6 to 0...30. So you can reduce gain much more than before. I tested and can state that values of 10...15 (regarding of preamp construction) works well. There should be nothing else affected (S-Meter works as before). Please test and report.

yo8can, 2017-02-04:
Thanks Andreas. I have updated to the actual firmware release, 04.02.2017, there are really  many enhancements and the mcHF works better in many ways (not tested all, I must learn the new menus etc.) and the filters are exceptional ! The feature MAX RX Gain works, but my first impression is that it acts like a variable attenuator inserted before the codec's inputs, and its internal gain stays always at maximum value, independent from this adjustment. This is not good because the noise generated by the codec remains the same and it is big...This setting affects the S-meter indications, in a manner similar to connecting two potentiometers between the OPA outputs and codec inputs, the bad thing is that in this case the internal noise given to codec remanins at the highest level independent to the MAX RX GAIN setting, which makes this adjustment usable only at 0...1...2 values and no more. I wish that the MAX RX GAIN should work by limiting the internal gain of the codec, in a manner similar how works the RF GAIN but not raising the zero indications of the S-meter, when RX Gain is lower. I suggest searching a method of adjustable lowering the gain of the codec in a manner acting similarly of controlling the gain of an operational amplifier, with a feedback resistor between inverting input and output, lowering the gain=lowering also the internal noise. It would be very good a feature like that, allowing a better gain balance between codec and OPA. More better, if such adjustment should be splitted at least one for lower HF bands, and other for bands over 18 MHz. I think that the existing MAX RX Gain setting cannot be used for Rx noise minimizing, because reducing it by increasing up to 30 this setting has the unpleasant effect of highlighting the internal noise of the codec against the signals.
Regards, 73, Romi

df8oe, 2017-02-04:
Hi Romi,

that was the only way to get influence via firmware. All other aspects are hardware-related, and I do not see any chance to improve something (except small improvements). S/N ratio is already "from the best" and cannot be reduced by firmware.

Again: I have worked many times using mcHF with a very small wire antenna and it satisfied me always.

Because of there is no firmware related issue I close this issue report. That does not mean that you cannot reply - it only means it vanishes from "open firmware issues" :)

BTW: Physics tells us that first stage of a receiver is responsible for S/N. So reducing of amplification later does not impact S/N. I think mcHF has already *very good* S/N....

73 de
Andreas, DF8OE

closed 2017-02-04 by df8oe

yo8can, 2017-02-04:
Hi, Andreas, 
That's we wanted to get, the noise of the receiver to be given mostly by the first stage...the BFR93...but this can be reached only by carefully balancing the gain in the Rx chain, else, like in my mcHF the noise of the Rx is given by OPA and codec, and NOT by the BFR which has nearly unsignificant contribution to RX noise. I'll try may best, but only in hardware...
73's, Romi YO8CAN

DD4WH, 2017-02-04:
Hi Romi,

I am conviced that you will not hear noise that is produced by the mcHF in your headphones, but all the noise comes from your antenna (especially for sure, if you receive below 10MHz). 

A good test is to switch off the AGC and listen to the receiver noise without antenna and then connect the antenna. If you hear more noise with the antenna connected, there is no point in thinking about noise anymore, because all the noise comes from your antenna ;-). Have you tried that on different bands? (AGC MUST be OFF for this test! Otherwise you will hear artifical noise that is not present in normal operation).

73 de Frank

df8oe, 2017-02-04:
You are not right. It *is* given by BFR - *not* by OPA - *not* by Codec.

73 de
Andreas, DF8OE

DD4WH, 2017-02-04:
Some good papers on gain and noise:
http://www.flexradio.com/downloads/qex_may2002-pdf/
http://www.arrl.org/files/file/Technology/tis/info/pdf/030304qex020.pdf
Have a look at table 6: below the 10m-Band, in nearly all situations the band noise will be higher than the mcHF internal noise ;-)

yo8can, 2017-02-04:
Thanks for the Iinks. I have tested the noise of the receiver without antenna, also by eliminating the BFR 93 (shorting its base to GND => Ic=0, BFR gain =0, the transistor=OFF, noise of BFR=OFF) the noise of the Rx doesn't lower significant (less than a dot in the bargraph), so the BFR's noise contribution is irrelevant here. Without antenna the AGC should set automatically the gait at maximum, even enabled. In absence of signals the ACG on or off, should make no difference. At my mcHF, the noise lowers only if the Gain of the OPA or codec lowers, the noise contribution of BFR in Rx is covered by the noisy OPA and codec. If the OPA stage can be improved, the codec remains the problem, it's noise is too big. If the noise of the BFR would not only in theory, but also in practice really give the internal noise of this receiver, without antenna, I would not open this discussion at all, being very satisfied. In other receivers, without antenna, If I turn off the first RF amplifier (if used), the internal noise of the Rx goes significant down (the AGC anyway been without antenna at maximum gain, like OFF), remaining only the lower noise contribution of the IF stages, mixers etc. (like in theory) but here, in my mcHF not, the internal noise is generated between the OPA amplifier and the codec, overwhelming the noise of the BFR.  I reffer only to internal noise, NOT the antenna's & band noise, this is the problem. Of course, I agree that, with a good antenna and good propagation the receiver is acceptable, but with small antenna mine is too noisy. 
73, Romi YO8CAN

DD4WH, 2017-02-04:
Hi Romi,
to be honest, I do not get your point exactly. 
You say that you have problems with noise generated in the mcHF. The noise figure of the mcHF has not been measured yet, but why should it be vastly different to other IQ SDR receivers?

How do you measure noise? Please do not measure it with the mcHF S-Meter. If you did that, which S-meter setting did you use? Old school setting is not at all reliable, in fact, it is dependent on AGC settings. Which bandwidth do you use for noise measurements? Always use the same bandwidth, otherwise the figures are not comparable.  
Also, there seems to be a misconception: "If I turn off the first RF amplifier (if used), the internal noise of the Rx goes significant down" --> How did you measure that? Most of the modern receivers have preamps with a noise figure of about 1dB (like all these modern MMICs). I do not think you will be able to hear that as a significant difference . . . 1dB difference is extremely difficult to hear ! 
So I assume what you are hearing is the AGC and NOT the noise per se, as would probably be the case in the mcHF.

Please also think about the difference between software and hardware gain. Software gain = AGC gain does not produce any noise. What do you mean by codec gain and how did you lower that?
You said you referred to internal noise and not band noise, but in which situation with which antenna do you experience internal noise that is larger than the band noise in your case "with small antenna mine is too noisy"?

Sorry to be so pushing, but I would really like to know how we can optmize the mcHF, but in order to do this it is very important to really try to focus on the questions and to get the things really clear and identify them very precisely.

73 de Frank

DD4WH, 2017-02-04:
also rediscovered a quote by Gerald Youngblood 2003 on the SDR-1000 (which has a very similar frontend as the mcHF):

"In fact, the noise figure of the analog front end is so low that if it were not for the atmospheric noise on the HF bands, we would need to add a lot of gain to amplify the thermal noise to the quantizing level."

http://www.arrl.org/files/file/Technology/tis/info/pdf/030304qex020.pdf

So we would need very good arguments in order to use better OP-amps, a better codec etc. for a better noise figure . . . ;-).
73 de Frank

df8oe, 2017-02-05:
Hi Romi,

Testing / comparing noise is only possible in AGC OFF mode, RFG and AFG staying at the same setting. If you leave AGC activated it will adjust its gain so noise is almost at the same level regardless of hardware conditions. Digitally noise AND signals are bits and bytes and cannot be divided by hardware or software. There are existing some DSP solutions which *can* reduce noise but these solutions have CPUs inside with 10 times and more horsepower and gigabytes of RAM - and their only task is sound processing (they must not drive a complete radio hardware as second task). It is NOT POSSIBLE to add such noise reducing DSP function to mcHF MCU firmware!

I never ever do have noise problems with all of my mcHFs and you are the first during the whole time I joined the project (about 3 years ago) who tells there is an issue. So it is a legal question:

Do hundreds of users not identify that there is a noise problem?

or

There is no problem at all?

Sorry but I think there is no problem at all. Some of your statements are physically "nonsense" (OP-amp produces noise...) and others I cannot confirm. You are not sitting at the same radio I am thinking...

Do you have another owner of a mcHF nearby so that you can compare yours to another? Maybe your special rf PCB does have a very strange and undiscovered issue?! But it is *not* a general problem at all.

73 de
Andreas

yo8can, 2017-02-05:
Hello,
How do I measure noise: Today I have evaluated the noise of the receiver by measuring the output AF voltage applied to loudspeaker, using a broadband (10Hz-1 MHz) AC Voltmeter, always at AFG=18, BW=2.3kHz, without antenna connected. So I can appreciate S/N ratio, measuring the noise audible voltage without any signal applied, compared to voltage increasing when a certain level of signal (from RF generator &ATT) is applied to antenna socket. E.G., for a 10dB S/N ratio the AF voltage must be approx. 3 times more with signal applied, then without signal present (only noise sound voltage) and so on, for other ratios.
To evaluate the codec's own noise introduced, I have disconnected its inputs (let them open) and turned OFF AGC (set to MANUAL). In this case, a 0.4Vrms of hiss appears to the loudspeaker, sounding very strong at RF Gain set to 37 (S1 indication). When raising the RF Gain more, the hiss becomes infernal, reaching 1.5Vrms over RFG 40 setting. At 50 I have not tried, making sure to not destroy the speaker or the 386.
A tolerable (high) level of hiss appears at approx 0.1V RMS on loudspeaker, this appears at a RFG setting of maximum 33 (S7 indication, not relevant do to known issue of classic S-meter here). Under this RFG number, the noise lowers, being acceptable at RFG less than 30 (S7 indication). This codec without signals at its inputs acts like a withe noise generator.
Observation: The codec becomes unstable at manual RFG settings more than 39 ! The noise raises suddenly at dangerous values for the AF amp and speaker. This never happen when AGC was set MEDIUM (ON, automatically). 
Further on I have remade completely the Rx path as the original schematic and using a signal generator and attenuator, with RFG OFF (manual) and made some signal to noise ratio tests.
- For 50uV injected to the antenna socket, (as should be at signal S9  not S-meter), obtained S/N ratio 50dB, very good ! (RFG=32, 1 Vrms versus 3mVrms without signal)
- For 5uV, S/N ratio was 31 dB (OK) and
- For the minimum possible using my test rig, about 1.6uV (like for S4 signals), the S/N ratio was 21dB, wich is good. According to those tests appears that there is no problem with the noise in my Rx. If the classical old school S-meter had been calibrated, than I had not been so in error.
Thanks Andreas and Frank for the support.
73's, Romi YO8CAN

df8oe, 2017-02-05:
Hi Romi,

nice measurements.... I think we can go a little bit better by improving preamplifier. But normally band noise is much higher than S4 so this has not "high priority". So we can finish discussion about noise coming from mcHF :)

73 de
Andreas

yo8can, 2017-02-05:
Hi Andreas, 
OK, Thanks for support and the great work shared.
73, Romi YO8CAN