df8oe/UHSDR #773: DTR for CW carier

WD8BXS, 2017-02-04:
Wow, this looks like a great add, I do not get it to work in fldigi, Where exactly do I set it up in the fldigi?
I have selected DTR for PTT but no luch.

Thanks,
 de Chuck WD8BXS

db4ple, 2017-02-04:
Hi,
it is NOT PTT via DTR it is CW Key via DTR, PTT is still only FT-817 CAT. The DTR line can be used like a straight key in order to enable / disable carrier (when TX was enabled by CAT PTT command).

WD8BXS, 2017-02-04:
I do not understand, Is this something that can be used in fldigi to send CW with the rig in CW mode? Here is a pic of my screen, I see no where to address what yo are talking about.
[image]

db4ple, 2017-02-04:
I have no idea how fldigi does CW. I have also no clue how rigcat works, never used it, only Hamlib. 
The request is based on the problem to get UCXLOG to work, not fldigi. 

BTW, in the current daily this function does not work at all correctly.

df8oe, 2017-02-04:
It is not a neccessary feature for most external software. Normally you can control mcHF via CAT commands (this works for SSB and CW). The best choice is to use "only CAT commands".

Helpers like "PTT via RTS" or "DTR" are from many years ago when CAT does not support switching PTT or doing CW via CAT.

73 de
Andreas

dm6tt, 2017-02-04:
As much as I know fldigi uses the right audio channel with a 1600Hz tone to do cw-keying.
http://www.w1hkj.com/FldigiHelp-3.21/html/cw_keying_page.html

DG9BFC, 2017-02-04:
ptt via rts would be a nice add on for digital mdes software that can not cntrol ptt via cat command
then any software that can toggle rts can be used with mchf

df8oe, 2017-02-09:
If it is ok now please close this issue.

closed 2017-02-09 by WD8BXS