df8oe/UHSDR #784: CAT Using wrong band power settings.

dalerobins, 2017-02-07:
If I used WSJT-X or HRD with CAT the mcHF will go to the correct frequency as requested.
One thing I noticed is if I am on one band for example 80m and  with CAT I choose a 40m band such as 7,076 the power settings are not correct, when using Power buttons on the mcHF if the VFO B is different to VFO A, the settings do not work correctly. It seems to be using the power settings for VFO B.

Also if you go in to the PA menu to adjust, it will show 40m but does not adjust properly. Because the radio will show Bnd80m on the "VFO B", even though it only lets you set the 40m band.

db4ple, 2017-02-07:
CAT will not change bands, it will change frequency (same as you can do, if you just keep changing the frequency). The bandlabel in the top left corner will tell you which band memory you are using. The band indicator right next to the frequency is derived from the frequency.
Since this may disrupt one of your band memories, there is a menu controlled switch to tell the mcHF to not store the CAT frequency in the band memory. Check the Wiki (Menu Item List).

PA Adjust: yes, you are right, we check the band memory id not the dial frequency. Let's say this is a minor problem. Please create a separate issue for this, and we will fix this in due time (not very high priority I'd say). 
VFO B vs. VFO A -> need to check that, should not happen. 

The code makes sure that the displayed power level (e.g. 2W) is applied according to the actual tx frequency. If the transmit is on the right frequency it should use the right values for that frequency.
The band memory name (Bnd80m) is not relevant.

So please give an exact procedure how to reproduce the problem. Include all settings (mode, vfo a and vfo b value, what exactly you expected to see and what you observed). Otherwise difficult to find.
BTW, whether you use CAT or not makes no difference in this area. The low level TX code does not know anything about where the "TX" command came from. So you should be able to reproduce the problem also without CAT. Think of CAT doing nothing more than you do when you turn the knobs and press the buttons.

dalerobins, 2017-02-07:
Ok that makes sense.

Sorry I was getting RFI in to the PC making it look like a lower power was coming out of the radio.

All is working fine, and  you are correct it's just PA adjust that doesn't work when you have sent it the frequency via CAT.

So you can only set the PA Adjust for that band if Both VFO show the same.

Thank you for your reply.

df8oe, 2017-02-09:
If it is ok now please close this issue.

df8oe, 2017-02-21:
close forced.

closed 2017-02-21 by df8oe