df8oe/UHSDR #791: when usin new agc (wdsp agc) the s-meter does not work

DG9BFC, 2017-02-12:
title says it all ... s-meter does not work ... plus i have problems with the setup ... signal seems overloaded (distorted in the peaks) if i tune in to a stronger am station ... ok ... turned down agc threshold .. now audio is better ... then tuned to a lower signal (not so strong station) ... almost unhearable ... used the "normal agc again and both stations are about the same strength in speaker (NO OVERLOADING FROM THE BIG GUN)
so ... seems as the new agc does not work like it should (or i do not know how to use it)

hb9ewy, 2017-02-12:
I also observed the S-meter issue, only with Old-school S meter, dBm based S-meter works, linearity ok.
No overload issues observed, possibly due to different settings. 
WDSP AGC Settings: threshold: 50dB, Max gain: 0, Med, slope 3dB. Daily from Feb. 11
At startup WDSP AGC Thresholod setting is always 0dB.
73, yves

DD4WH, 2017-02-12:
hi guys,
thanks very much for your testing and issue reports!
You are too quick for me, this hidden feature has not even yet been announced on the forum . . . 
For now, as a quick fix (very similar to what Yves has stated):
Use WDSP AGC Settings: threshold: 60dB, Max gain: 0, slope 6 to 10dB and as you like FAST/MED/SLOW/LONG/VERY LONG, hang: OFF (but try HANG ON with different settings: different hang times implemented, experimental!)

Settings are not yet saved in EEPROM.

old school S-meter only works in standard AGC mode, because it is 100% dependent on the AGC variables in the standard AGC code. Will have to think more how to fix that.

I will prepare a detailed explanation and implement more adjustments. I will post that in the German forum and in the github-Wiki in English. But AGC is not the simplest feature on a radio, so please have some patience, needs some more days. AGC is a feature that needs patience in understanding and testing, until we have adjusted it to our/your specific needs and to the specific band conditions.

Please wait until the announcement on the forum with more issue reports ;-). Do not close the issue, however, we will probably need it . . .

DG9BFC, 2017-02-13:
solved

closed 2017-02-13 by DG9BFC