df8oe/UHSDR #796: Predefined filter selection by G4 not working

yo8can, 2017-02-13:
Hello,
After upgrading to 04.02.2017 and also actual 12.02.2017 FW, predefined filter selection by short pressing G4 button doesn't work any more, only long pressing enables the filter selection by Encoder 3. Before, using an old firmware (middle of 2016 year), worked ok the filter selection using G4, so I think the hardware must be OK. Please advice,
Regards,
Romi YO8CAN

df8oe, 2017-02-13:
Hi Romi,

I think you do not have configured filter sets for the different modes in filter menu...

Please try this and when it works for you close this issue.

73 de
Andreas, DF8OE

yo8can, 2017-02-13:
Thank you Andreas,
After configuring filter sets in filter menu it works fine.
73, Romi YO8CAN

closed 2017-02-13 by yo8can

df8oe, 2017-02-13:
Hi Romi,

thank you for your feedback.

Software enhancement is very fast so that firmware which is older than 2 weeks almost is "outdated" ;) So in your case newer firmware needs configuration.... That is possible at other places, too.

Good luck and have fun using mcHF

Andreas

yo8can, 2017-02-14:
Hi Andreas,
Thank you for the great support.
The MCU fitted in my mcHF is STM32F407VET6 and the EEPROM is 24LC1026. Please advice me what MCU and EEPROM should I buy in order to work with future firmware versions. Thank you.
73's Romi YO8CAN

df8oe, 2017-02-14:
Hi Romi,

serial EEPROM is the one we recommend. But MCU is not. It only has 512KB flash
memory and firmware size is already quite near to this. I recommend to use MCUs
with 1MB flash. And if you want to swap MCU it would be a very good idea to take one
which has more RAM, too. We are not sure if this will become relevant in future - but
price is only 50 Cent higher so we recommend this, too.

These do have 2MB flash and 256KB RAM:
STM32F427VIT6
STM32F429VIT6
STM32F437VIT6
STM32F439VIT6

73'2
Andreas

yo8can, 2017-02-14:
Thank you Andreas.
73, Romi YO8CAN