df8oe/UHSDR #799: Transferring callsign to the mchf

steinhj, 2017-02-14:
Is it possible to transfer some static variables from the usb stick in to the mchf during firmware upgrade? 
( for example reading a file at the usb-stick, - "static.txt" ,  containing Call=callsign)
It could i.e. be used to send callsign identification in freeDV.
(and in welcome-screen)

db4ple, 2017-02-14:
No. 

Not possible due to lack of software for doing that. We are actively working on a solution for that general problem (only the bootloader is able to use the large usb connector).
However, for that specific problem there maybe a different solution (also not working yet). We started to implement the Clone Protocol of the FT817. Part of this Clone Protocol is the Callsign. You will be able to set the callsign in Chirp (or any other tool which supports the FT817 Clone Protocol) and send it to the mcHF for use. But this is limited to a call sign only (and is only partially implemented).

73
Danilo

steinhj, 2017-02-14:
Thanks for a very quick ansver, - and for your excellent work !

I close this request

closed 2017-02-14 by steinhj