df8oe/UHSDR #801: LINE UT STILL DISTORTED

DG9BFC, 2017-02-15:
title says all
line out signal is still "scratchy" even if i turn down line out level
tried down from 80 to below 30 and sound is still distorted
test can be done best with an aktive speaker on line out
if i use speaker out sound is much clearer (up to output level of 20 or so)
distortion seem to be a few stages before the audio level control (on line out) cause it is still there even if i turn down the level
to have clean sound on line out i have to set the wdsp agc around 20 db lower (but then the agc light is not blinking like it should so agc is working below the normal threshold)
PLEASE fix this distortion on line out

df8oe, 2017-02-15:
NO DISTORTION CONFIRMABLE

Tested using active PC speakers connected to line out.

DG9BFC, 2017-02-15:
hmm ... seems as wdsp slope is too low with default setting to 4 db ... above 6db signal sounds better ...
3 db or below and signal is very distorted ... ok ... i have to play a bit more with the settings ... maybe find better settings with more listening and testing ... i will report back and when i find good working setting i will close issue

closed 2017-02-15 by DG9BFC