df8oe/UHSDR #812: Power adjusting less than FULL POWER doesn't work at FW 1.5.9

yo8can, 2017-02-18:
Hi, 
Today I have installed the 2.0.1 Bootloader, and used the big USB socket to load the latest Firmware, 1.5.9. using a TDK 8G memory stick and worked OK (the only that worked for me with the V1 bootloader & mini USB socket, with the 2.0.1 bootloader not tested other memory stick).
Everything was ok untill testing the transmit power, on a dummy load, using the TUNE push button. The transmitter delivered the full power over 10W on all bands disregarding the power selection (0.5W. 1W...5W) so I try next to adjust the power on all bands again from menus. The Full power adjustings are the only ones that worked, I stored the settings for 10W on all bands. Next I try to adjust the drive levels to get 5W but didn't work, when selecting a band and the menu for 5W adjusting, when pushing the TUNE button to transmit and adjust the power for 5W, the Tx Bias text gets white (not the selected power adjustment parameter) and could't adjust the power for 5W any more.
After saving these settings to EEPROM, when selecting any power from 0,5 to 5W the power on the dummy load is now 10W on all bands, could't adjust them using the menu.
Please advice,
Regards, 73, Romi YO8CAN

DG9BFC, 2017-02-19:
there is a menue setting for the tune power
that can be set to 0.5w or 5w or to "same as tx power"
only if you set that to same as tx power you can change the powersetting while using tuine button
if you wanna be able to change the power settings with tune you have to change that
with also selecting that power level ... means for 80m 5w can only be changed when you also have selected 80m band and set also the power to 5w
why you have same level on all settings?? easy
you stored same level for full and 10w and 5w ... mchf does calculation to 0.5 with calculating the curve from 10 to 5 to 0.5 ... but when you set both the same there is no difference between them so then calculation can not work 
do the setup again and it will work .... power setting in the different bands and modes are long solved
(remember cw bias can only be set in cw mode etc.)
so romi .. .good luck ... you sure will solve that (when doing again that settings)
dg9bfc sigi

df8oe, 2017-02-19:
Here is everything working as expected.

Setting TUNE POWER to 5W results in 5W tune power, setting to 0.5W results in 0.5W power. Pay attention: tune power only can be changed in RX mode...

yo8can, 2017-02-20:
Thanks Sigi and Andreas,
Now it works here too.
73, Romi YO8CAN

closed 2017-02-20 by yo8can