df8oe/UHSDR #822: idea RTC via I2c module?

AngelSerra, 2017-02-24:
Can the RTC be implemented via i2c?. For example the module that is used in arduino, chip use DS1302, DS1307 or modern DS3231 with eeprom at24c32, that only one would have to read the address of the eeprom since it is kept the hour the day the month the year the day of the week,to reduce the noise of the i2c the cpu would only read for a certain time the eeprom address alone, few modifications would be needed on the board to implement this type of rtc as it has battery and the mchf off the rtc would still work with just a simple reading When closing the mchf.

db4ple, 2017-02-24:
Hi,
of course, you can do that, but why, it you just need to follow this mod:

https://github.com/df8oe/mchf-github/wiki/Recommended-Modifications#ui-0405-n-031-enable-stm32-real-time-clock

df8oe, 2017-02-25:
I see same way. We already have a very good running solution included software correction . So there is absolutely no need to implement new ones. BTW: the EEPROM which is present on this modules do disturb existing EEPROM on mcHF so you have to modify the modules....

s53dz, 2017-02-25:
I fully agree. The RTC as implemented here using an internal STM's RTC works really good. I do not miss any other function here.

AngelSerra, 2017-02-26:
Yes, via software only no backup batery

df8oe, 2017-02-26:
?? No backup battery? What are you talking about??

db4ple, 2017-02-26:
The implemented STM32 RTC is using the RTC hardware inside the STM32 and a battery as described in the link I provided. It runs 100% independent from the firmware (same as it would be with an separate i2c module) AND it needs a battery, just like the external module.
And as Andreas pointed out, the modules with I2C EEPROM are not good at all, due the the EEPROM possible interferring with the existing one (if the RTC is connected to the same I2C bus).      
Please read the description of the mod again and get back if you have questions.

Thank you!

closed 2017-02-28 by df8oe