df8oe/UHSDR #825: Not possible to set RTC after reset

wemaus, 2017-02-26:
With last FW from 25.02.17, 14:54:10 it is not possible to set RTC after reset of RTC. There are only 0 and 1 for time settings.

db4ple, 2017-02-26:
Did you do "RTC Start" (i.e. is the RTC being displayed and running)?

db4ple, 2017-02-26:
Ok, you may forget my last question, you only see the time change entries if you did RTC Start.
What was the last working firmware for you? Is the time running (I know you can't set it but does it count the seconds etc.)?

wemaus, 2017-02-27:
Yes I do, but nothing has happend (No RTC encountered). So I have installed an older FW and start RTC again. After that I reloaded the last FW.
I don't think that there is a problem with my oscillator.

db4ple, 2017-02-27:
We did not change much after the initial RTC implementation so it is strange the that new one does not work for you. Which version (build time) was the firmware which worked for you?
Knowing this would allow us to narrow done the issue probably much quicker.

Thank you!

wemaus, 2017-02-27:
The FW before was 11.02.17 11:02:28 which run with RTC first time. With this FW I had installed the RTC.

AsbjornPettersen, 2017-02-28:
i had a comment on the patch on "MchfRtc_SetPpm(int16_t ppm)" function lately.
Don't know the rest of the code, but it looks strange. Maybe it can cause problems !?

db4ple, 2017-03-02:
No, MchfRtc_SetPpm(int16_t ppm). Or at least extremely unlikely. No one else reported an issue with the mentioned firmware. I think it is clearly a usage problem. After executing "RTC Reset", the RTC is completely disabled (not even shown on the display anymore). It is not possible to set the RTC in this state and shows the above mentioned behavior. You need to restart the mcHF, run "RTC Start" (as described in the Wiki  https://github.com/df8oe/mchf-github/wiki/Recommended-Modifications#ui-0405-n-031-enable-stm32-real-time-clock) and then the clock can be changed again after the automatic reboot.

Please close the issue, it is not a bug. Will update the wiki to include more explicit mentioning of the "RTC Reset" purpose.

closed 2017-03-04 by df8oe