df8oe/UHSDR #833: sync am (sam-l and sam-u) has a bug

DG9BFC, 2017-03-13:
i can stop mchf working with procedure as follows:
switch to sync am
with long press switch to sam-l (or with another to sam-u
then short press and mchf is stopped completely (no audio, no display change, no power button works so you have to kill power supply (pull the plug)
...
if after long press and switch to sam-l i do another long press ... and another ... then no problem with pressing short
but when in sam-l or in sam-u a short press kills mchf
when in sam or am ... no problem
only in saml and u the problem exist

DD4WH, 2017-03-14:
Hi Sigi,
can´t reproduce that here. Which version of the firmware are you using?
Is it a HAL version or the regular version and which date?
73 de Frank

df8oe, 2017-03-14:
It is only reproducable on HAL version.

db4ple, 2017-03-14:
It only happens if you switch to FM, if you disable FM it does not happen. Will look into the interaction between SAM & AM and FM Demod and Mode Switching.

DG9BFC, 2017-03-14:
i reflashed the versin from 5th of march (pre hal version) ... there the problem does NOT exist
 ... when using pre hal version all is well
when in sam-l or sam-u ... a short press of mode button switches to fm
when i use the hal version it has the problem described above
i will use the older version till you have found that bug ... then i can switch over to hal version

db4ple, 2017-03-14:
Hi,
with my modified HAL source code I cannot reproduce the error anymore (before I could). I reorganized the switching very little but it seems to have done the job. Not sure that I fixed the issue in general (since I don't know what fixed it, there was nothing really suspiscious).

Have to trace it a little more. It is for sure not strictly HAL related, I suppose it is related to stack/memory usage.

DG9BFC, 2017-03-14:
can you send me the bin for trying it out??
or send a commit and andreas can make a new dayly snapshot??

db4ple, 2017-03-14:
Hi Sigi,
now I am feeling better, I can reproduce the issue even with my new changes (which I could not before as mentionend). If I switch magnification levels,  the problem is there. It is not the magnification level, it is the switching. Also if I start in magnification x4 I can reproduce the problem. More investigation necessary.

DG9BFC, 2017-03-15:
seems solved
with first try i got "red digits" ... but after that it worked fine
so seems to work like the non hal version (you can switch from sam-l or sam-u to fm without killing function)

commit 535f716d (2017-03-15): Fixed #833 (stop if switching from SAM-L/SAM-U)
Was a race condition during filter switching as it seems
closed 2017-03-15 by DG9BFC