df8oe/UHSDR #835: Freedv 1600 works fine ... can we implement 700c mode??

DG9BFC, 2017-03-17:
Title says it all
i just listened to the pa guys in mode freedv (they meet nearly every day on freedv mode) ..... after i asked them for switching to 1600 mode i could copy them very fine here in northern dl
...
but they are also trying the 700c mode (that needs fewer signal level with same voice quality as the 1600 mode)

could we (you programmers) implement also the 700c mode?!?

db4ple, 2017-03-17:
Hi Sigi,
I am afraid, this is not going to work. With 1600 we are close to the performance limits and 700C is most likely beyond. Of course we can (and should) try, but I am not very optimistic here.

szjdb, 2017-03-18:
Yes, it is very difficult as the author of FREEDV still have not done that in his SM1000. The RAM size and horse power is far from needed. Hope rewrite some of the code by your MCHF team! Good job!

df8oe, 2017-03-18:
I do not think this is possible using small STM32F4. The maths which are needed cannot be modified to fit in mcHF (which already runs a SDR software). We will wait if this mode arrives SM1000 and if so, maybe we have a look at it if we are able to reduce horsepower and RAM which is needed to fit in parallel with SDR code.

closed 2017-03-18 by df8oe

DG9BFC, 2017-03-18:
hmm ... freedv 1600 needs only 21-22% cpu load in mchf ... while analogue ssb needs 41-42%
i am guessing that implementation of the 700c code in sm1000 is not done yet cause the 700c code is under development (first tested on pc systems and doing over the air tests) ... there is another new mode 800xa (if i remember right) ... so it should be (maybe!) possible to implement one of the newer (fdv) modes when the testers find out what to use (700c or the new 800xa mode)
maybe another mode?!? ... so keep on hammering on the keyboards ... grin
when freedv development is close to finished .. i am sure the clever programmers can also add that to the mchf code in parallel to the sdr, display etc.

db4ple, 2017-03-18:
Hi Sigi,
that freedv uses only 21-22% is a misunderstanding of the Load indicator. The load indicator shows the load of the AUDIO Interrupt processing only. This is where most of the audio processing is done in all non-digital modes. FreeDV uses a different approach and does the computation outside of the AUDIO Interrupt. This results seemingly low load but trust me, the FreeDV 1600 processing is close to 90% sometimes. It was very hard work to get it to work anyway (meaning to get it below 90% reliably).

DG9BFC, 2017-03-18:
ohhh ... ok ... that are details that i did not know ... i was thinking load represents the complete stm load ... ol then ... back in ma black hole