df8oe/UHSDR #840: Error in include.mak

AsbjornPettersen, 2017-03-22:
Backslash "\" on last line in include.mak will break the mingw build.

drivers/usb/device/class/composite \

Should be:
drivers/usb/device/class/composite

db4ple, 2017-03-22:
Can it be fixed by adding one more empty line in the Makefile after the include instead of changing this file?

Reason is that I use a simple script to generate the files from the eclipse builds (which adds a backslash to every file name.

AsbjornPettersen, 2017-03-23:
Yes it works with GNU Make 3.81 !

od -c include.mak 
0002400  \n   d   r   i   v   e   r   s   /   u   s   b   /   d   e   v
0002420   i   c   e   /   c   l   a   s   s   /   c   o   m   p   o   s
0002440   i   t   e       \  \r  \n  \r  \n
0002451

AsbjornPettersen, 2017-03-25:
Thanks. Latest patch work fine. Have other problem with MinGW build, but havn't found the fix. 
It's another issue.

db4ple, 2017-03-25:
Windows Makefile builds will not work with current makefile due to too many files on command line. Only bootloader build works due to fewer files. So try bootloader build. If this works, but the other does not, command line lenght limitation is the problem. Moving some files in a library solved the issue for Eclipse but this adds a hefty 70k more of flash usage. Okay for my internal development with an 1M flash but not okay for offical releases which should run on 512k machines as well.

AsbjornPettersen, 2017-03-25:
Thanks for the information, now i see why "make all" and mchf fails to build.
The bootloader build, but only doing a "make clean", "make bootloader".
This is because "make all" will build main.o, stm32f4xx_it.o, ui_lcd_hy28.o and system_stm32f4xx.o without the -DBOOTLOADER_BUILD and cause a link error on "make bootloader". Same on Linux !.
A temporary fix will be to do a clean on the 4 files before building:
preclean:  
	$(RM) $(call FixPath,basesw/mcHF/Src/main.o)
	$(RM) $(call FixPath,basesw/mcHF/Src/stm32f4xx_it.o)
	$(RM) $(call FixPath,drivers/ui/lcd/ui_lcd_hy28.o)
	$(RM) $(call FixPath,basesw/mcHF/Drivers/CMSIS/Device/ST/STM32F4xx/Source/Templates/system_stm32f4xx.o)

Another problem is the hex2dfu.py is written for Python 2.x and i have Python 3.x. I'll try tp port it.

db4ple, 2017-03-25:
Hi,
you have to build in separate build directories:
E.g. 
mkdir bl; cd bl; make -f ../Makefile ROOTLOC=.. clean bootloader

See travis.yml in the mchf-github folder 

Please remember to provide the port in a different file so that we can run both old and new python

AsbjornPettersen, 2017-03-25:
The Python 3 port for hex2dfu.py stopped by:
from compat import asbytes, asstr  ->ImportError: No module named 'compat'

Then i found a hex2dfu for C, and it was easily build with gcc.  This can maybe replace Python 2 code.

df8oe, 2017-03-26:
Because we are using different platforms (Win, Mac, Linux) I want to stay on python code...

AsbjornPettersen, 2017-03-26:
I totally agree.   A Python3 port should be Plan B,   C code Plan C :)

AsbjornPettersen, 2017-04-01:
Trying to get arount the Windows build problem (too many files) ) by makeing libraries, but i get many errors.  Any ideas for the AR options ?

libsrc.a: $(LIBOBJS)
	$(AR) cru $(MACHFLAGS) -flto --specs=nano.specs   $@ $(LIBOBJS)

db4ple, 2017-04-01:
Check what Eclipse does when it runs "DebugLib" builds, you will see how it calls ar

AsbjornPettersen, 2017-04-04:
I found a workaround to build "make all" on windows and msys/MinGW . I reduced the number of c files from 502 to 439 and it builds. Looks OK.

arm-none-eabi-size mchf.elf
   text    data     bss     dec     hex filename
 425168    1808   90468  517444   7e544 mchf.elf

df8oe, 2017-04-05:
Hi Asbjorn,

this is an isue in Windows/msys/MinGW - not in firmware or Makefile. Reducing files is a particular solution but will pop up some time later again, when number of files increases. I think you should write an issue report to MinGW developers and use a Linux based machine if you want to build using Makefile in the meantime. If you do not want use Linux use Eclipse: that works well on Windows.

AsbjornPettersen, 2017-04-05:
I found a sollution. If i reduce the foldername (ex FilterFunctions to ff) then the MinGW/msys works! So it's a buffer issue and not the number of files. Havn't  configured the windows version of Eclipse correct, but i'm not sure it will work with latest build system.  No problem to use Linux, though

db4ple, 2017-04-05:
Hi,
as mentioned in a previous comment, Windows has a command line length restriction which is relatively low compared to Linux. Eclipse works fine under Windows, no problems (except for the command line length issue, which is also there but handled by the way we build under Eclipse, also mentioned in a previous comment). 
The best way is not to change the directory names, but to find a way not to use the directory names at all in the gcc link command line. This is possible, if all .o files are collected in the same directory. Which again only works if the names of all files are unique (which they are right now except for main.c but this could be easily fixed). 
Changing the directory names (or file names) would break some workflows for the HAL part (integrating updates from STM would be much, much more complicated). So this is not recommended.

closed 2017-05-11 by AsbjornPettersen