df8oe/UHSDR #844: Problem loading mchf.bin with new bootloader

Haasoperator, 2017-03-24:
I upgraded my mcHF v. 0.4 to bootloader 3.0,00, and updated the F/W (March 17) via USB cable and DFUse Demo. However, when I tried to update to the newest daily snapshot (March 22) using the  same USB thumb drive I used with the old bootloader, it does not work, and neither did the other thumb drive I tried (and the new bootloader is SUPPOSED to recognize ALL thumb drives now). I have heard at least one other person on the Yahoo group say he could not get a thumb drive to work with the new bootloader, either.

db4ple, 2017-03-24:
Hi,
thanks for the report. Not much we can do about it I am afraid without more information. Could you please give us the error code of the bootloader when used with your drive. Easiest way is to flash bootloader 3.0.4, which will use the display to tell you the problem but bootloader 3.0.0 is fine, it uses red led flash groups. Please count the flashes in one group and report back.

If you just see a green blinking led, your drive is not recognized at all. 
BTW, it is really recommended to use the newest bootloader 3.0.4 , we have fixed a few handling issues with the buttons.

73
Danilo

Haasoperator, 2017-03-24:
Problem fixed with 3.0.4. Now it tells me I have to keep BAND- pressed to update the F/W, rather than simply holding down the POWER button. MUCH better. Closing the issue.

closed 2017-03-24 by Haasoperator