df8oe/UHSDR #845: TouchScreen reverse can not save.

BI3MEK, 2017-03-26:
bootloader 3.2
fw 2.2

73

df8oe, 2017-03-26:
I cannot confirm. Works for me (using serial EEPROM). Do you have installed serial EEPROM?

73

BI3MEK, 2017-03-26:
I use 24cm01

db4ple, 2017-03-26:
Hi,
how did you find out that the reverse is not saved? Key Test Screen?

BI3MEK, 2017-03-26:
1. Set “Reverse Touchscreen”  ON
2. Touch normal
3. Press the POWER button to turn off the machine
4. Turn on
5. “Reverse Touchscreen” display ON,but Touch screen the x direction to reverse.

73

db4ple, 2017-03-26:
Thanks, that helps. Other settings such as receive frequencies are stored at power off (and restored when starting)?

BI3MEK, 2017-03-26:
Other settings are normal。

db4ple, 2017-03-26:
Hi,
I'll attach a new firmware binary to this comment. Please try this. Should fix the problem (if I am right, that is).

[mchf.zip](https://github.com/df8oe/mchf-github/files/870820/mchf.zip)

BI3MEK, 2017-03-26:
ok!!The problem fixed. thank you!

73

db4ple, 2017-03-26:
Ok, will push this fix to the repository and it will appear in the next firmware builds.
Thank you for reporting the issue!

BI3MEK, 2017-03-26:
I will close this issue.

closed 2017-03-26 by BI3MEK