df8oe/UHSDR #847: No Audio in firmware 2.0.0+

williamhemmingsen, 2017-03-26:
I have no audio/spectrum/waterfall etc. in firmware 2.0.0+.  I am using ui/rf board version 6.  Firmware version 1.6.0 works perfectly

db4ple, 2017-03-26:
Hi,

please run the attached firmware and check what the System Info menu says regarding Audio Codec (yes or N/A).

[mchf.zip](https://github.com/df8oe/mchf-github/files/870958/mchf.zip)

73
Danilo

williamhemmingsen, 2017-03-26:
It says "Audio Codec Presence    **Yes**"

db4ple, 2017-03-26:
Well, that makes it somewhat more complicated. I need some time to get an idea what to do next.  
If you happen to have an I2C EEPROM, you can try to solder it (recommended anyway). This would be a clear shot into the dark. 
If in TX mode: is there any audio from external sources shown (set Meter to AUD, set Input to L>L, MIC or L>R), and use an external to send audio to the mcHF. You can try this on 1.6.0 to see what the effect is, and then repeat on the newest firmware.

Thanks!
Danilo

williamhemmingsen, 2017-03-26:
Tried mic on both firmware versions.  the AUD meter only registers on v1.6.0.  Nothing on the new build.

db4ple, 2017-03-26:
Thanks. No surprise here.
We have no audio in, we have no audio out. Only logical explaination is a problem in the I2S stream setup/operation and/or I2C setup (but we check at least basic functionality here). So this leaves us with the I2S stream setup/operation.

We know that we don't get any audio data from the codec, we also don't hear any data sent towards the codec.... how can we check if there is any data sent out from the cpu.

williamhemmingsen, 2017-03-27:
I have a EEPROM on order,  We will see if that makes a difference.

df8oe, 2017-03-27:
Problem 100% identified. It is the missing serial EEPROM. It is mandatory to use serial EEPROM but we will fix firmware , too so that audio comes up even without EEPROM.

db4ple, 2017-03-27:
Hallo Andreas,

meine Versuche sind nicht erstmal zielführend, vergiss den Pull-Request. 
Ich melde mich heute abend...

73
Danilo

williamhemmingsen, 2017-03-27:
Audio now works in latest daily.  So I will close this issue.  Thanks again!

closed 2017-03-27 by williamhemmingsen