df8oe/UHSDR #853: UiLcdHy28_Font()

AsbjornPettersen, 2017-04-05:
Found some dangerous code:
If BOOTLOADER_BUILD then legal index will be 0, otherwise 0,1,2,3,4
if the bootload code call with "font" index 1,2,3,4 it will point outside the table!
Can't see any illegal "font" use today but it's easy to do so later on.

ui_lcd_hy28.c :

static sFONT *fontList[] =
{
#ifndef BOOTLOADER_BUILD
        &GL_Font8x12_bold,
        &GL_Font16x24,
        &GL_Font12x12,
        &GL_Font8x12,
        &GL_Font8x8,
#else
        &GL_Font8x12_bold_short,
#endif
};

const sFONT   *UiLcdHy28_Font(uint8_t font)
{
    const sFONT   *cf;

    if (font >4)
    {
        cf = fontList[0];
    }
    else
    {
        cf = fontList[font];
    }

    return cf;
}

------------------------------------
Quick fix:

const sFONT   *UiLcdHy28_Font(uint8_t font)
{
    const sFONT   *cf;
#ifdef BOOTLOADER_BUILD
    cf = fontList[0]; /* only one font ! */
#else    
    if (font >4)
    {
        cf = fontList[0];
    }
    else
    {
        cf = fontList[font];
    }
#endif
    return cf;
}

db4ple, 2017-04-06:
Well spotted.

Thank you.

AsbjornPettersen, 2017-04-06:
Great. You also reduced the number of code lines,

df8oe, 2017-05-07:
if this is finished please close issue.

closed 2017-05-07 by AsbjornPettersen