df8oe/UHSDR #855: structural programming

AsbjornPettersen, 2017-04-20:
In misc/serial_eeprom.c :

I found some code in  serial_eeprom.c that's a little difficult to read.
When investigating the function it break some rules for structural programming.
Not a big thing, but the code will be difficult to read and easy to make mistakes.
This function is not a problem, but if there are thousands of code, it will be hard to maintain.

static void SerialEEPROM_24Cxx_AdjustAddrs(const uint8_t Mem_Type, uint8_t* devaddr_ptr, uint32_t* Addr_ptr);

If we call this function twice with the "same" parameters the output will differ.

static void dummy_example()
{
    uint8_t devaddr;
    uint32_t Addr = 0x10000; /* example */
    uint8_t Mem_Type = 17;

    SerialEEPROM_24Cxx_AdjustAddrs(Mem_Type,&devaddr,&Addr);
    /*----------------------------------------------------*/
    /* devaddr will be set to (MEM_DEVICE_WRITE_ADDR + 8) */
    /*----------------------------------------------------*/

    /* NB! Addr have been changed ! */
    SerialEEPROM_24Cxx_AdjustAddrs(Mem_Type,&devaddr,&Addr);	
    /*---------------------------------------------------*/
    /* devaddr will be set to (MEM_DEVICE_WRITE_ADDR ) ! */
    /*---------------------------------------------------*/
}

I'll try to explain the problem (see Addr_ptr):

SerialEEPROM_24Cxx_AdjustAddrs() parameters:
const uint8_t Mem_Type, /* control flow(Ok) */
uint8_t* devaddr_ptr,   /* data output(good !) */
uint32_t* Addr_ptr,     /* control flow(Ok), data input(good), data output(Not needed!) */

The original code:
static void SerialEEPROM_24Cxx_AdjustAddrs(const uint8_t Mem_Type, uint8_t* devaddr_ptr, uint32_t* Addr_ptr)
{
    /* data output(good !) */
    *devaddr_ptr = MEM_DEVICE_WRITE_ADDR; 

    if (*Addr_ptr > 0xFFFF) { /* control flow(Ok) */
        switch (Mem_Type) /* control flow(Ok) */
        {
        case 17: // 24LC1025
  	    /* data output(good !) */	
            *devaddr_ptr = MEM_DEVICE_WRITE_ADDR + 8;            // 24LC1025
            break;
        case  18:
	    /* data output(good !) */
            *devaddr_ptr = MEM_DEVICE_WRITE_ADDR + 2;            // 24LC1026
            break;
        case  19: // 24CM02
	    /* data output(good !) +  data input(good) */
            *devaddr_ptr = MEM_DEVICE_WRITE_ADDR + ((*Addr_ptr & 0x30000) >> 15); // the upper bits 16 and 17 determine the I2C address offset
            break;
        }
	/* data output(Not needed!) */
        *Addr_ptr &= 0xFFFF; // mask address to 16bits in all cases
    }
}
The 16 bit address(Addr_ptr) will be masked anyway later on.   /* eeprom_desc_ptr->addr = (uint16_t)((0xFFFF)&(Addr)); */

More structural code:  
NB! const uint32_t Addr_ptr, /* control flow(Ok), data input(good) */ 

static void SerialEEPROM_24Cxx_AdjustAddrs(const uint8_t Mem_Type, uint8_t* devaddr_ptr ,const uint32_t Addr_ptr)
{
    *devaddr_ptr = MEM_DEVICE_WRITE_ADDR;

    if (Addr_ptr > 0xFFFF) {
        switch (Mem_Type)
        {
        case 17: // 24LC1025
            *devaddr_ptr = MEM_DEVICE_WRITE_ADDR + 8;            // 24LC1025
            break;
        case  18:
            *devaddr_ptr = MEM_DEVICE_WRITE_ADDR + 2;            // 24LC1026
            break;
        case  19: // 24CM02
            *devaddr_ptr = MEM_DEVICE_WRITE_ADDR + ((Addr_ptr & 0x30000) >> 15); // the upper bits 16 and 17 determine the I2C address offset
            break;
        }
    }
}

static void dummy_example_structural()
{
    uint8_t devaddr;
    uint32_t Addr = 0x10000; /* example */
    uint8_t Mem_Type = 17;

    SerialEEPROM_24Cxx_AdjustAddrs(Mem_Type,&devaddr,Addr);
    /*----------------------------------------------------*/
    /* devaddr will be set to (MEM_DEVICE_WRITE_ADDR + 8) */
    /*----------------------------------------------------*/

    /* NB! Addr have NOT been changed ! */
    SerialEEPROM_24Cxx_AdjustAddrs(Mem_Type,&devaddr,Addr);
    /*----------------------------------------------------*/
    /* devaddr will be set to (MEM_DEVICE_WRITE_ADDR + 8) */
    /*----------------------------------------------------*/
}

db4ple, 2017-04-20:
Hi,
yes. Not the best way to solve the problem. However, as you pointed out, not a real problem, since always used correctly. Best way to help others is to completement this comment with a pull request with a better implementation.  You are invited: improve code, test, make nice pull request, we review and your code is in.

AsbjornPettersen, 2017-04-23:
Ok. i'm very into test driven development, so i like to test the code before adding it into production code.
I have download the google test and made one unittest just to se how difficult it would be.
Made one unittest where one test failed and one passed. I incuded one mcHF c file into the linux test program, and it worked. How usable it is in a "hardware" environment it's another question.

closed 2017-05-11 by AsbjornPettersen