df8oe/UHSDR #858: Error handler loop

AsbjornPettersen, 2017-04-29:
I'm looking into the bootloader code and i see that the bootloader.bin file size is 31782 and i assume that the EEPROM size are 32768 (0x8000). 986 bytes free.

When i look into the code i see that there are some "copy & paste" code, and i think i could reduce the code by 1-2 KB.
See baseSW/src/tim.c,,,
Shall i improve the code ?

The error handler looks unfinished. Should be a blinking led or delay. 

void Error_Handler(void)
{
  /* USER CODE BEGIN Error_Handler */
  /* User can add his own implementation to report the HAL error return state */
  while(1) 
  {
  }
  /* USER CODE END Error_Handler */ 
}

db4ple, 2017-04-30:
Hi,
files under basesw/mcHF are generated using STM32CubeMX. Most files are copied from standard libraries except for basesw/mcHF/Src. These are generated depending on the configuration. We try to limit changes to these parts in order to be able easily regenerated files. We had to break this in a few places but the general rule is to keep the hands of this code. 

Now for the proposed improvements (deduplication). I am all in for it in general. But this is not going to help the case "bootloader size". Why? Because we use link time optimization, which removes all not referenced code very well. Since the bootloader does not use the TIM routines, no code from this file is being include, so changes would bring no benefit. 
Have a look at the generated bootloader.map file in order to see what is effectively being used in the "real" image.  We use SPI, DMA, USB Host, FATFS and GPIO from the basesw. USB and FATFS are the biggest contributors to size, have a look by yourself.

ErrorHandler -> yes. we could and maybe should do something here. Although this Error_Handler should never be called in normal operation (it is used only during machine init) and it should never trigger in a "normal" mchf firmware as it can only fail on STM32F4 internal peripheral init (not the external). During firmware development you may wrongly initialize a peripheral and can trigger the error handler. Since we rely on STM32CubeMX to do the configuration right it is very unlikely to happen.

AsbjornPettersen, 2017-04-30:
Ok. Great to have a good linker.
But why compile and link unused code in the first place ?

These files should be removed from bootloader.mak 
basesw/mcHF/Src/tim.c 
basesw/mcHF/Src/adc.c 
basesw/mcHF/Src/i2c.c
...

df8oe, 2017-05-07:
Bootloader length is also dependent on used compiler. I have many different compilers stating from 4.9.0 parallel on my development machine and can select via environmant variable which one I use for building. So I test at new releases of bootloader which one gets the smalles binary. Bootloader is at actual state a few bytes smaller as 32.768 bytes (of course).

Please close issue if it is solved for you.

closed 2017-05-07 by AsbjornPettersen