df8oe/UHSDR #859: Enhancements for portable use

om1aeg, 2017-05-01:
Hallo, after SOTA & OMFF activation with mchf trx powered from 3S LiPo Battery i have idea for LOW BATTERY warning (configurable voltage level with some delay, sound or text on LCD) and switch off at configurable voltage level, also with delay.
Is it possible add this feature?

Also readability of LCD on direct sunlight is bad, maybe inverted colour schema or switchable normal / remove / minimize some items on LCD together with bigger frequency and S-Meter can helps in portable, Maybe dynamic LCD with 2 screens, one "basic" when touching LCD, knobs, buttons and one after few seconds when no "touch something on radio" with bigger frequency, S-Meter, waterfall, but without "level" , "mode" "input" items. 

Configurable lock is another enhancement for portable use.

Thanks, Anton, OM1AEG

df8oe, 2017-05-02:
Actually there is nothing planned regarding this. If you use 3.2" LCD and adaptor PCB brightness and contrast are better than on 2.8" LCD.

yo2ldk, 2017-05-04:
Referring to display, can be added waterfall + spectrum option, or this overload the STM ?

df8oe, 2017-05-04:
Overload of CPU and too low pixel amount for a useful view.

yo2ldk, 2017-05-04:
and if we use SPI LCD mode to free I/O, this will be possible?
https://www.seanet.com/~karllunt/S4DVGA.html

DG9BFC, 2017-05-10:
i think a user setable low batt or "undervoltage" alarm should be easy
and would be nice for users that have a built in battery pack
maybe "blinking backlight" as prewarning??
and/or a text that pops up (low battery alarm switching off in 10...9...8...7)???

...
we already have the voltage reading on the screen ... you just have to look on it when you are doing portable ... (but no preventing of deep discharge of a li ion pack!!!)

phaethon, 2017-06-08:
+1 vote for undervoltage alarm/shutdown with two level configurable thresholds.

phaethon, 2017-06-27:
I have created a simple implementation, which:
- adds configuration setting "Low power shutdown" under Configuration menu
- setting variates OFF - 9.0, 9.1, ..., 12.0V
- if not off it shuts down MCHF if voltage is below the threshold

Adding a delay and prior notification could be useful for someone, but I found that I am often operating with LCD off when from battery. I would not notice visual notification. Adding a notification sound might be disputable for others.

I can send a diff or compiled binary if anybody wants to test. Going to send a pull request later.

om1aeg, 2017-06-28:
phaethon,
Thanks for the support and implementation.

When is mcHF in RX mode, then is function without delay ok, can save the LiPo battery when I forget turn off the radio on the table.
But when is mcHF in TX, then in SSB - full power - voice peaks drops voltage for short time. Can be to solve with HW modification (serial resistor - parallel capacitor in measure input) or with SW delay.

phaethon, 2017-06-28:
@om1aeg  Automatic shutdown is not performed while in TX mode. Also, shutdown check is performed only after voltage value update, which creates additional delay. Did you test the fw and it shut down right after tx in an inappropriate situation? What delay do you suggest after switching from tx to rx?

om1aeg, 2017-06-28:
@phaethon 
When is "power off" active only in RX, then ok. I must check function "in real work" on adjustable Power supply.

"What delay do you suggest after switching from tx to rx?" Two second is OK, I'm activated tens of SOTA or WWFF from LiPO battery (3Ah and 5Ah) and voltage go up very fast after end of TX (observing on FT817 / FT857 display).

I see in schematics, supply voltage is measured via divider R13, R14 with parallel C31 (10nF). Then, if is needed, HW smoothing (delay) is possible via change C31. But this affected voltage measurements delay when RX to TX and also TX to RX.

phaethon, 2017-06-29:
@om1aeg , please, check the latest fw with low power shutdown feature "in real work" to see if TX->RX delay is actually needed. It is easy to add it into the code.

om1aeg, 2017-06-30:
I do some practical tests with 3Ah 3S LiPo. When is on power Input connector 11,6V in RX mode, then is in worse case (FM full power 12W) voltage between 11,1 and 11,2V. In SSB is much better, between 11,2 and 11,4V (normal voice operation).
With the latest fw 30.6. 7:37 is "low voltage power off" working in RX mode after 50-55 seconds. Before mcHF shut down is blinking value of voltage. 
In TX is "LV power off" not working, voltage value is blinking, but radio still transmits (3 minutes test), but after end of TX radio directly shut down.
I'm also play with 2 channels scope and delay between low voltage in TX and higher voltage in RX is up to 1 seconds. Im also found delay between push PTT and key TX around 110mS, the same delay after release PTT.
Function "low voltage power off" is usable without problem. Maybe "blinking" of voltage value may be better with faster blinking.
Thanks for your time and work on this.

db4ple, 2017-06-30:
Hi,
a) thanks for the test!
b) Making blinking faster requires some more work (we use the voltage display update frequency as clock, this is =2.56s so rather slow)
c) I think we should change to not to immediately switch off after TX (indication should happen, though) since undervoltage measure is not only caused by discharge state but also by cable resistance in TX case due to higher currents. 
d) Off-topic TX delay: The delay is caused by a number of different things: 
- use of scope/spectrum may slow down reponse a little
- Depending on the input signal channel, there is a minimum delay for going to TX and you can configure this delay to be longer than the fixed minimum in the menu. Digital audio in (USB audio) and CW should have much less delay.

phaethon, 2017-06-30:
@om1aeg , what threshold value did you use when testing?
I am going to do a couple of additional tests throughout next week using LiFePO4 battery, which might have a different behavior than LiPo. It would be interesting to do another test with lead acid battery, but I don't have one.

phaethon, 2017-06-30:
@db4ple may be additionally to voltage text blinking, we could blink the green led when power is at the 'orange level'? That could be noticed with LCD off, too.

db4ple, 2017-06-30:
Sure, we could do this. Makes sense. However, we have to disable the load indicator in FreeDV (uses the green led, too) first since this would interfer with the blinking. And would save a little power (since for about a minute the led is 50% off :-) )

phaethon, 2017-06-30:
I have no idea how important is LED usage for FreeDV load indication as I am not active FreeDV user (yet). We could either disable low power behaviour of led while in FreeDV mode, or change behaviour once power is getting low or make it configurable (if led indicates low-power or load). From ui perspective my guess is that uninitiated user would intuitively associate led more with power issue than load.
Also, it could start blinking when at 'red' or 'orange' level (.5 up from threshold). Lots of options ;) but the more I think about the more I like the idea as such. Somebody actively using FreeDV should comment on this.

db4ple, 2017-06-30:
I don't think anyone will need the FreeDV LED, but some will notice.
I don't think we should blink before reaching the desired level.

But something else crossed my mind: I am not sure, if "low power shutdown" is a good menu item name.
It does not tell exactly what this shutdown is related to. How about "Low Voltage Shutdown" or something to that effect? Same applies to Low Power Threshold then to link both.
Although we have the nice menu help generated, I am quite sure most will not look at it regularily and would ignore the setting and leave it at its default (off).

Second thing is: we should set the reset default voltage to something sensible like 9V (similar to the old hard-coded voltage levels).

phaethon, 2017-06-30:
@db4ple , will you implement LED blinking or shall I attempt?
I agree that "Low Voltage Shutdown" is more clear.
I am not sure about good default settings though. If majority of users have 13.8V supply (my assumption) and shutdown setting is off, 12V could be sensible default for color notification (in line with former color notifications), too. For somebody enabling the auto shutdown, voltage would be manually adjusted anyway (possibly even regularly changed if using different batteries).

db4ple, 2017-06-30:
You can do the led thing. Please do not forget to rebase/pull the newest state from github.

There is a function to turn on/off the green led (see mchf_board.h) . There is now also pwmt.undervoltage_detected which is true  as long as undervoltage is detected.
The proper place to control the blinking is the UiDriver_MainHandler, where you could regularily updat the led state. Keeping this in a central place will allow us to use the led for various (low speed) indications. 
I will take care of removing the FreeDV Led code.

DG9BFC, 2017-07-15:
we have now the low voltage shutdown ... maybe issue can be closed??

closed 2017-07-29 by df8oe