df8oe/UHSDR #860: MChf error

oz9hi, 2017-05-02:
mchf issue 4 problem?
Transmitter dont Work while the codex  output from the mic. (Lout and Rout) not let it out.
Setup the power is not possible, you cant change the numbers.
Processer and codex are replaced.
I hope you can help me.

BR Jorgen oz9hi

df8oe, 2017-05-02:
This cannot be a bug because of hundreds other mcHFs do work properly. So it is an operational problem.

Please try first adjusting power using "TUNE" function. Values can only be changed at TX on specific band in specific power! Normally they are "not changeable". Adjust all power settings for every band in FULL and 5W position.

oz9hi, 2017-05-02:
Thanks for quick answer
As mensient  it is not possible to chance the power numbers, its lockt
BR Jorgen

oz9hi, 2017-05-02:
By the way the TUNE button when prest is red

df8oe, 2017-05-02:
Cannot confirm. Nothing is "locked" :)

So step by step:

1) switch to 80m and tune to a frequency within allowed frequency range (mcHF only transmits in ham bands)
2) set mode to LSB or CW
3) set power to "FULL" using button G3
4) in PA CONFIGURATION set "Tune Power Level" to "as TX PWR"
5) now in PA CONFIGURATION select "80m Full PWR Adjust" (the number is WHITE)
6) now press short "TUNE" and adjust power level

I just tested at my mcHF - is working properly.

oz9hi, 2017-05-02:
Thanks its ok now
Next problem
Noisereduction on, transmitter on and of, - an 3v signal from codex LHPout to speaker amp, and no more sound. Reset and all is ok Again.
I think its a known failure, what do you mean?

Thanks for your help
73 oz9hi@io.dk 
Jorgen

df8oe, 2017-05-02:
Yes its a known problem. NB is eating so much horse power that mcHF sometimes stops working. Also it does not reduce noise, so do not use it :)

closed 2017-05-02 by df8oe