df8oe/UHSDR #866: VFO A + B Tracking Enhancement Request

WD8BXS, 2017-05-08:
I use VFO A for CW and VFO B for SSB most days, except when I need to work spllit.

It would be nice that when I switch to VFO A or B, it would remember the filter settings that I was using.
That doesn't sound too hard for you geniuses hihi.

Thanks,
Chuck WD8BXS

df8oe, 2017-05-08:
It is not hard for us - but very hard for all users... Because of storage structure in EEPROM is hard coded it is not possible to add a new storage byte for doing that without loosing all your individual settings. So we do not want to implement such things until we start to implement completely new storage structure (for using memory places for ~100 "stations"). So users do loose all settings only ONE TIME ;)

73
Andreas

WD8BXS, 2017-05-08:
I understand Andreas, thanks for considering it!!

73, Chuck WD8BXS

From: DF8OE 
Sent: Monday, May 08, 2017 12:46 PM
To: df8oe/mchf-github 
Cc: WD8BXS ; Author 
Subject: Re: [df8oe/mchf-github] VFO A + B Tracking Enhancement Request (#866)

It is not hard for us - but very hard for all users... Because of storage structure in EEPROM is hard coded it is not possible to add a new storage byte for doing that without loosing all your individual settings. So we do not want to implement such things until we start to implement completely new storage structure (for using memory places for ~100 "stations"). So users do loose all settings only ONE TIME ;)

73
Andreas

df8oe, 2017-05-10:
This is a planned feature in the future. I close this issue but it is remembered!

closed 2017-05-10 by df8oe