df8oe/UHSDR #877: Waterfall AND spectrum

yo2ldk, 2017-05-21:
will be nice if we can have on screen in the same time waterfall and spectrum, and no"frizzing" display on TX
   here is an interesting project , also with STM
[https://www.youtube.com/watch?v=mrf9BsNrIHc](url)

and source files:
[https://yadi.sk/d/QuyGdEwM3H4HcB](url)

DD4WH, 2017-05-22:
Yes, looks very nice! And what a nice highspeed display!
I am not sure how that is done in the Malamute SDR. Maybe they have a whole processor dedicated to the spectrum/waterfall display alone and a second STM32F4 doing the DSP? Or they drive their display by DMA?
At the moment, that would be very hard to achieve in the mcHF from my point of view . . .

db4ple, 2017-05-22:
Frank,
the full source code is available in a zip somewhere on the page from the second url. Having a brief look at the code, I found they do a few "tricks":
- Run at 216 Mhz (outside spec) 

use@mcHF:
Hm. Probably possible. Just need to set all the clock parameters accordingly. 

- Use a SPI DMA screen buffer for the spectrum portion (SPI DMA blasts out the whole spectrum without blocking the CPU. 

use@mcHF: Without FreeDV it would be no problem but FreeDV eats memory like hell. However, all devices with more memory would clearly benefit here. It would make touch a little more sluggish, since we share the same SPI. 

Still one full screen update takes 28ms@216Mhz according to a comment I found in the code. 
Not bad but with about 75% Interrupt load at FreeDV and some SyncAM situations, we would go down to less than 10 updates per second even if we do could do it that fast.  SSB is not as demanding. 
On the F7 we should have not much trouble to get it working, we have more RAM and we have much more horsepower to spend.

DD4WH, 2017-05-22:
Hi Danilo,
thanks for the details! I also had a brief look at the code, but was unable to interpret those things.
Yes, I see that the spectrum/waterfall display "eats" at least another 31k of memory . . . 
However, they have proven that DSP Fast Convolution filtering in the frequency domain (FFT - complex multiply - inverse FFT) is feasible with the STM32F4, nicely done.  
Yes, we should keep that kind of display type and handling in our minds for the F7.

df8oe, 2017-05-25:
From my view LCD space which is available is too small for showing both. There is no proof in such a small part containing both...

df8oe, 2017-06-30:
please close this issue or comment again

closed 2017-07-14 by df8oe