df8oe/UHSDR #879: new agc causes filter ringing (pling pling sound)

DG9BFC, 2017-05-22:
title says it all
the old agc turned the gain down on every peak (i have some electric fences here ... grmbl) that was not good (mchf became deaf between those peaks with slower agc)

new agc behaves a bit better but it causes "ringing" of the filters

a 2.3 kc filter rings at 2.3 kc and a 2.7 kc filter rings at 2.7

the lowpassfilter ringing more as the bpf filters

the wider filter ringing less then the narrower filter

the wider bpf filter sound more "natural" (peak sounds like a normal noise peak)

klack klack klack ... sounds better as pling pling pling :-)

maybe that can be a bit finetuned??

df8oe, 2017-05-25:
Hmmmm..... I cannot reproduce... Maybe DSP NR or NB is enabled?

DG9BFC, 2017-05-25:
no ... i did not use nb or nr ... can be best heard with the 1.6 kc lpf ... but also the other filters have that

and it is also with old agc (but a bit lower so i did not notice before) ...

it can be heard if you have an electric fence or similar noise .. only then! ... and during the last months the fences here were not in use (only under power from spring to autum ...

since a few days i know have those fences under power and so i hear the filters ringing

df8oe, 2017-05-26:
I do have fence here and I cannot reproduce. So no chance to fix it. We must wait if others report this behaviour, too under different circumstances...

df8oe, 2017-06-30:
This is normal behaviour if you do have very sharp filters. No possibility to get BOTH: sharp filters AND no ringing. Goal was to improve suppression of wrong sideband - so you have to live with ringing...

closed 2017-07-14 by df8oe