df8oe/UHSDR #881: odd use of boolean variable

AsbjornPettersen, 2017-05-23:
++,-- on  boolean variable !?

typedef struct AudioDriverState
{
    bool af_disabled; // if TRUE, audio filtering is disabled (used during filter bandwidth changing, etc.)
..
}

void RadioManagement_SetDemodMode(uint32_t new_mode)
{
..
  ads.af_disabled++;
..
 if  (ads.af_disabled) { ads.af_disabled--; }
..
}

Should be:
void RadioManagement_SetDemodMode(uint32_t new_mode)
{
..
  ads.af_disabled = true;
..
 if  (ads.af_disabled) { ads.af_disabled = false; }

db4ple, 2017-05-23:
Well,

af_disabled is not exactly used like an boolean variable. It is a guard for a critical path. We should change the type to uint16_t (or any other integer type). 
The purpose of the counting is to be able to have multiple functions to be able to prevent use of audio filters. Some of this functions may call other functions which in turn also are able to disable audio filters. Only after we left the last (outermost) function which disabled audio filtering, we re-enable it. That is the theory behind it. BTW, the type 'bool' is in fact a define which resolves to  'int' so this is why it works.

AsbjornPettersen, 2017-05-24:
Well, the ts.dsp_inhibit seems to do the "same" job:

  if  (ads.af_disabled) { ads.af_disabled--; }
  if (ts.dsp_inhibit) { ts.dsp_inhibit--; }

 uint8_t dsp_inhibit;				// if != 0, DSP (NR, Notch) functions are inhibited.

db4ple, 2017-05-26:
dsp_inhibit is something else. Also audio driver, but for something else (the noise reduction code).

closed 2017-05-30 by AsbjornPettersen