df8oe/UHSDR #882: DialFrequency findings

AsbjornPettersen, 2017-05-24:
struct DialFrequency  in radio_management.h :

ulong   update_skip;  # UNUSED!, just remove it
ulong   step_new;  // Eth driver req step  # Set, but not used. just remove it

  uint8_t dial_digits[9];
  uint8_t sdial_digits[9];     // Second display

Those tables seems to hold previous digits on the LCD display !?
They shall skip unnecessary digit display, but i can't se how it could work.
All the non zero digits will always be displayed.

ui_driver.c :
void UiDriver_UpdateFreqDisplay(,,)
{
...
 // -----------------------
 // See if digit needs update
 if ((digits[idx] != dial_digits[idx]) || ts.refresh_freq_disp)
 {
    bool noshow = idx > last_non_zero;
    // don't show leading zeros, except for the 0th digits
    digit[0] = noshow?' ':0x30 + (digits[idx] & 0x0F);
    // Update segment
    UiLcdHy28_PrintText((pos_x_loc + pos_mult[idx] * font_width), pos_y_loc, digit, color, Black, digit_size);
  }
..

Should be:
 if ((digits[idx] != dial_digits[idx]) || ts.refresh_freq_disp)
 {
    bool noshow = idx > last_non_zero;
    // don't show leading zeros, except for the 0th digits
    digit[0] = noshow?' ':0x30 + (digits[idx] & 0x0F);
    // Update segment
    dial_digits[idx] = digits[idx];  // Save current digit, to prevent unnecessary Lcd display
    UiLcdHy28_PrintText((pos_x_loc + pos_mult[idx] * font_width), pos_y_loc, digit, color, Black, digit_size);
  }

db4ple, 2017-05-25:
With the display speed improvements we removed the performance optimization of not changing digits since it did not handle the color changes. But we may want to re-add it later. So we left the data structure intact.

df8oe, 2017-06-30:
please close or comment again.

df8oe, 2017-07-06:
Hi Asbjorn,

we have added "Hall of Fame" and I put all contributors I remembered to it. Of course you are listed, too. But I do not know if you da have a callsign? If so: Please add (or tell me) - or do you not wish to be listed?

closed 2017-07-29 by df8oe