df8oe/UHSDR #884: RTC-Battery wrong detection at turn on

DK8XK, 2017-05-25:
I have the RTC mod. completed. Works already, but the automatic battery detection plays a trick on me.
By accident, the battery is detected or not at start up, which leads to have no time and push buttons not available. The Ubatt. is 3.04 or 3.31 volts without changing the situation. If I turn off and on again, the battery is detected and I can get the clock start and so on some power-ups later. This is unsatisfactory in the long run.  I suggest to remove the automatic, and simply turn the battery on or off via menu please. Any comments? vy 73 de Willi, dk8xk

db4ple, 2017-05-25:
The automatic detection does not directly measure a voltage. It detects the ability of the backup RAM to store a value even when the processor is off. The backup RAM works down  with Vbatt 1.8V and less. So my guess is that there is a hardware issue with your mcHF. Doublecheck that there is absolutely no connection between VBat pin and Vcc (3,3V digital). Then check the battery to MCU connections.
73 Danilo
BTW, the automatic detection is not perfect but important to keep the mcHF usable. An undetected running RTC blocks the use of the mcHF keys F3 and M1 (they are considered to be pressed all the time) so you would not be able to change the setting in the mcHF...

DK8XK, 2017-05-25:
Dear Danilo, yes, I double checked this. Event, if mcHF is ON or OFF, the voltage at MCU's pin 6 ist identical to batteries voltage everytime. This is why I feel it's a software question to have sometimes the wrong battery test what ever piece of software will do this. Any idea? vy 73 de Willi

db4ple, 2017-05-25:
Please report used firmware and bootloader versions. The software simply reads a specific location in the backup RAM and checks if it is still the one the software wrote in there last time. If this fails, it can have 2 different causes:

a) The VBat supply is not working properly on you machine, causing the backup RAM to be reset. 
b) The software changes that value somewhat randomly. 

 We can almost rule out b), since if this would be the case, more reports would come in. Software would do this for all users. 

Leaves us with a) as most likely cause. It is true, that we may have a software bug causing that problem but there has to be some more evidence for that. We will take note and wait. Try identify some more details when this happens.
73 Danilo

df8oe, 2017-05-25:
I am nearly sure this is a "personal hardware issue".

DK8XK, 2017-05-25:
Dear Danilo, FW:2.2.9 May 14.2017, BL:3.2.0. I switched more then 25 times on/off no issue. I had the situation that I switched on 3 times with no detection. The 4th succeeded. Battery's voltage apply constanly with no interuption to MCU's pin 6. The device is not moved, no earthquake etc. happens. If the FW runs well on other machines, then the only HW which cause the issue may the MCU itself? vy 73 de Willi

db4ple, 2017-05-25:
So lets go into the details:

What exactly happens if is not detected?
You had the RTC running (and displayed on screen)
You switch on and now what?
RTC disappears -> what is shown in the System Info menu for Backup Battery?

DK8XK, 2017-05-25:
"You switch on and now what?"-->RTC runs normal 25 times with PSU constantly on. 
Then:
Ten minutes later I switched on and no battery. Switch on again and now battery. Nothing different compared to the situation before with 25 times no issue. I don't touch anything exept switch on the PSU and mcHF. But in other try's yesterday I switched both and no issue. And an other day I switched on in the morning and the RTC was still running. A bit confusing? Ah.. and I remember onetime I start the RTC in the "System Info menu" then the mcHF starts itselfe and ??no battery??. Start manually and battery is back. Isn't it funny?
"RTC disappears -> what is shown in the System Info menu for Backup Battery?"-->n/a
Next start-->yes
vy 73 de Willi

df8oe, 2017-05-26:
Maybe there is a spike when you switch on your PSU and that spike can cause changes in some bits (randomly)  of static RAM. So the contents of the static RAM can become unreliable. To test that try to work with PSU always "on" (even in power down state of mcHF). So no spikes can be generated...

DK8XK, 2017-05-26:
Tried with accumulator --> same as before.
If the RTC-mod is done it need only one time to tell the FW that RTC shall run. If I role back the RTC-mod then I will switch off via menu. 
Suggestion: If ON then "Start RTC, if OFF then "Stop RTC", thats all. 
PLEASE

DK8XK, 2017-06-11:
OK, solved. Add a 0,1µF ceramic condenser (low leakage current!) across Li-Battery and works stable from that moment on. Please add this new C82 to the mod: UI-04/05-N-031 Enable STM32 Real Time Clock. Vy 73 de Willi, dk8xk

closed 2017-06-11 by DK8XK