df8oe/UHSDR #897: Build problem ROM limit

AsbjornPettersen, 2017-06-06:
On Linux and latest code:
make clean
make all

i'll get:
region `rom' overflowed by 36848 bytes

arm-gcc-link.ld:
MEMORY
{
	rom  (rx)  : ORIGIN = 0x08010000, LENGTH = 512k - 64k
	ram  (rwx) : ORIGIN = 0x20000000, LENGTH = 128k
	ram1 (rwx) : ORIGIN = 0x10000000, LENGTH = 64k
}

I change ROM size and mchf.bin links again:
	rom  (rx)  : ORIGIN = 0x08010000, LENGTH = 1024k - 64k

Are 512k the real limit on f4 ?

db4ple, 2017-06-06:
Hi,
yes and no. The CPU originally sold with kits from M0NKA was a 512k variant. So for those, 512k is the limit. For that reason we try to keep the firmware below this size. 
It depends a little on the compiler settings and the actual compiler if we are below 512k. So the official build is below that margin. If we change back to O2 in the Makefile as default, your build will be also below 512k-64k again (or use a different compiler).

AsbjornPettersen, 2017-06-07:
The -Os will optimize for size (bootloader only)
The -O3 will optimize for speed (default for Linux)

In EmBitz build used -O2 for "release" and it build for 512K ! And for "releaseSmall" it uses -Os
So there could be an extra small build of mchf.bin for the 512K version on Linux !?

closed 2017-06-09 by AsbjornPettersen