df8oe/UHSDR #906: LCD white !!!

ik2fwy, 2017-06-13:
HI all
 LCD White
Config HW Pcb 05 LCD HY28B Spi or parrallel mode, White screen
CPU STM32F407VGT6TR new ram 24LC1025
The board seems to respond to commands band +- vol adjust, Tune, on, off, is ok
bootloader df8oe ,
firmware df8oe via USB stik all ok
restart.... White screen
help me

df8oe, 2017-06-13:
Actually there is no known issue "white screen". Informations you provided are very low - so I guess:

1) LCD widely out of specs or faulty (can you test using another LCD which works confirmed)?
2) Hardware problem (bad soldering)?

Also interesting to test using a much older firmware - 1.x versions...

db4ple, 2017-06-13:
Hi,

first of all I would check all the connections on the 0.5 board being correct for the selected display mode.
There is plethora of 0R resistors to be placed correctly for each of the modes. See DF8OE Wiki for information regarding the pinout. If you bought a presoldered board from Chris, it is configured for parallel mode.

73
Danilo

db4ple, 2017-06-13:
What I forgot to say, as far as we know, most problematic displays which expose a white screen in parallel mode seem to work ok if configured properly in SPI mode (and the resistors on the PCB are configured accordingly)

ik2fwy, 2017-06-13:
Ciao Danilo
ho preso solo i PCB da Chris, senza componenti  i componenti li ho presi da
farnell e digikey ho comperato 2 lcd!

credo che sto facendo un errore, no le resistenze non le ho modificate
 dove trovo le modifiche da fare mi hai già mandato un link ma no le trovo
non so quali sono
questa descrizione riguarda solo il touchscreen ho anche la parte grafica
lo schema da adottare dove si trova

UI 0.5 HY28B/HY28A

If the touchscreen SPI is enabled, i.e. R30 - R32 are fitted, it does not
matter on wich side you connect the wires from the HY28B/HY28A pins.
Otherwise you have to identify the pad which goes to the MCU and not to the
display.

   - connect R30 (side to MCU) to LCD pin 16-HY28B /5-HY28A (SDO)
   - connect R31 (side to MCU) to LCD pin 17-HY28B /6-HY28A (SDI)
   - connect R32 (side to MCU) to LCD pin 13-HY28B /3-HY28A (SCK)

This will be used in future to get some more GPIO pins for further
enhancements. There are big advancements in the speed of SPI mode in the
last months. This change is (not yet) necessary.

UI-05-N-026 Activate touchscreen (recommended to use actual FW features)

Touchscreen implementation on board 0.5 is incomplete and in some places
not compatible with the firmware. On board 0.5 you must modify the ui board:

   - *fit* R30...R32
   - *remove* R33 and R34
   - *don't fit* R47b and R47d
   - put a wire between R33(side to LCD) and R47d (side to MCU)
   - put a wire between R34(side to LCD) and R47b (side to MCU)

mi dispiace che non riesco a farlo funzionare e darti disturbo

ho provato con multimetro ho continuità tra il display e le resistenze

73 Lorenzo ,attendo consiglio

ik2fwy, 2017-06-14:
Ciao Andreas  finalmente ho fatto funzionare il Display  !!!!
gli altri problemi li vediamo poco alla volta

Grazie Lorenzo

2017-06-13 21:58 GMT+02:00 Lorenzo Curti <ik2fwy@gmail.com>:

df8oe, 2017-06-14:
Ciao Lorenzo,

ciò ar il problema?? 

73
Andreas

ik2fwy, 2017-06-14:
Edit x lcd SPI now works thanks

I have to make the calibrations

ik2fwy, 2017-06-14:
Il manuale di M0NKA è diverso dal nuovo firmaware non mi trovo con il menu
73
Lorenzo

2017-06-14 20:15 GMT+02:00 Lorenzo Curti <ik2fwy@gmail.com>:

df8oe, 2017-06-14:
https://github.com/df8oe/mchf-github/wiki/Operating-Manual---Menu

73
Andreas

closed 2017-06-14 by df8oe