df8oe/UHSDR #921: CW memories and CW keyboard

ok1cdj, 2017-06-28:
It will be nice to have some memories for CW operation. Record memory from paddle or (i don't know if it is possible) connect usb keyboard to mcHF for  memory editing. Also send CW from keyboard will nice feature.

73 Ondra OK1CDJ

df8oe, 2017-06-28:
This is a far-in-future-feature. It is a "nice to have" of course. But not simple at all and takes much time and man-power. Its priority is not so high - but it is now on the wish-list :)

73
Andreas, DF8OE

pa7n, 2017-07-01:
Andreas,
Would CW memory without a keyboard easier to implement? Instead of the keyboard you could just dial in the characters with the encoder knobs. Not as easy as with a keyboard but workable I think.

73,
Erwin PA7N

df8oe, 2017-07-02:
It is much work, too. If we implement selecting characters via ancoders we have to implement selecting characters via touchscreen, too (it is much more comfortable than using encoders). I think it is nearly the same R&D so better wait until USB activation using keyboard will bei implemented. That is perfect solution. Or start it by yourself - it is already on 2do list.

73, Andreas

pa7n, 2017-07-02:
A qwerty keyboard on the touchscreen :) That would be fun!

ok1cdj, 2017-07-03:
Insert memory by  encoders is also ok for me. You don't need change it very often.
I am not fan of touch screen and i have no touch screen in my 0.4.

73 Ondra

df8oe, 2017-07-03:
Hi Ondra,

you may think so about touchscreen - but we must look at all users! And many users like touchscreen very much (included me). It makes many operations very intuitive, fast and simple and mcHF would not be the same radio when there is no touchscreen... So encoders is bundled with touchscreen. Or support via USB-Keyboard or USB Key (textfile).

73
Andreas

ok1cdj, 2017-07-03:
Hi Andreas USB key is also nice idea...

phaethon, 2017-10-04:
@ok1cdj have you checked out the cw keyer functionality committed lately? See description at the end of Digimodes section in the wiki: https://github.com/df8oe/UHSDR/wiki/Digimodes

closed 2017-11-04 by df8oe