df8oe/UHSDR #929: CW transmit frequency marker in bandscope

pa7n, 2017-07-02:
Just draw a line on the transmit frequency on CW so it's easily to zero beat a CW station.

Another possibility is to display the real transmit frequency exactly in the middle of the scope. As it is now the transmit frequency seems like it's in the middle of the screen but it's offset with the CW tone frequency.

Example: 10.116 as transmit/receive frequency, centre line of the bandscope says 10116.00 but the CW station appears 700 Hz on the left of the middle line. It should be on the middle line. In reality frequency on the centre line of the bandscope is not on 10116 but on 10116.7 MHz.

(in the cw settings menu I have "CW Freq. Offset" on shift)

73,
Erwin PA7N

df8oe, 2017-07-02:
Hi Erwin,

there have been many discussions of the "correct" showing of TX and RX frequencies in CW. And many different statements (and every statement was correct under a special view). So you can select between TX, RX and SHIFT in CW menu. In very old firmware there were much more settings which lead to confusion.

I am happy that I have understood the implementation which is now active and it works similar to all other (commercial) rigs ;)

73
Andreas, DF8OE

pa7n, 2017-07-02:
Hi Andreas, I don't have a ic-7300 so I don't know how that transceiver displays the bandscope on CW.
But I think it's not much discussion. I transmit on 10116. So I dial 10116. (as with any commercial transceiver) The centre of the bandscope should be on 10116. And if I receive a station it should be on 10116 in the bandscope because his carrier is exactly on 10116. Simple. The real transmit signals are leading here. Not any sidetone offset or the receive LO offset. 
If you use a commercial transceiver and dial in 10116 mode CW it's transmitting on exactly that frequency not on 10116.7 or 10115.3. So that 10116 carrier signal should appear on the bandscope on 10116 and not on +700Hz or -700Hz.
73, Erwin

df8oe, 2017-07-02:
And now discussion starts again :)

If I want to receive / hear a CW signal which is on exactly 3.600KHz I must tune to 3.600750 (if I am working in CW-L) or to 3.599250 (if I am working in CW-U). If I tune to 1.600 I do not hear anything.

So there must be a calculated sidetone in RX or TX frequency. Because of if I use my commercial rigs I never have had done any thoughts about this the discussions started when I tested the different modes old mcHF firmware offered. Most of them are absolutely useless because you are not "transceive" when you want to answer an incoming CW call.

We have had reworked this section 1.5...2 years ago and now again I must not think about anything. If I hear a CW station I can answer and I will be heard, too. So everything is correct and working fine. I remember hours and hours with dicsussions what is the "correct shown frequency" - uuughhh!!! ;)

73, Andreas

df8oe, 2017-07-02:
I have tested again:

I set mcHF to CW and 3.600KHz. I set my FT890 to 3.600KHz and CW, too. Now both transceivers can hear each other giving nearly the same sidetone frequency. So if I tell another station QSY 3.555 I *will meet him at 3.555*...

73, Andreas

pa7n, 2017-07-02:
It's not the problem of not hearing each other. The problem is the display in the bandscope. So tuning on frequency works perfectly. It's a bandscope problem! Got it?

df8oe, 2017-07-02:
Maybe it is coming up :) Do you mean waterfall and/or scope "receiving bar" is not on correct position?

pa7n, 2017-07-02:
:) I must be telling it way too overcomplicated! Sorry :) But yes that is in fact the problem. The bandscope is 700 Hz (or the side tone frequency you selected) shifted.

phaethon, 2017-07-03:
If I understand correctly what is being said by OP, I do agree. I have CW-U, CW Offset set as SHIFT, sidetone at 600Hz, filter e.g. at 600Hz/300Hz. When the main freq dial shows 7.010600 the numbers on the scope under the red needle show 7.010600, too, while it is the frequency, which should be displayed below the sidetone (i.e. the center of underline, which shows filter bandwidth is shown) and it should be 7.010000 below the "needle".

s53dz, 2017-07-07:
Hi,

I suppose to clear things out it is only necessary to add some clear lines to Wiki to tell exactly what the frequency display settings in menu mean.

For me this issue is really a simple one. Actually not really an issue. Originating from the modulation characteristics it is always only one frequency to show. Other are just subjective ones. And I agree that this only one frequency should be consistently showed in digits and on the spectrum/waterfall.

I attached here a .PDF table showing the right display frequency for some of modulations we use in HAM radios. Hopefully with this to close the issue.

73 Bojan

[Dial_freq.pdf](https://github.com/df8oe/UHSDR/files/1130513/Dial_freq.pdf)
(I also added it to my Files tab.)

df8oe, 2017-07-29:
Is it correct now? If so, please close issue.

s53dz, 2017-07-31:
Just a tiny report:

I just did the comparison between ver 2.4.0 and the latest 2.5.27.
In both versions I get the same display.

I get the same frequency shown (the actual TX frequency) in numbers as at the spectrum analyzer dial (needle) position. Which is right.

But I suppose this issue was only about the **position of a little horizontal line bellow the needle** in CW mode. This line is properly offset for the LSB/USB mode but in CW it is still offset by the BFO offset. For helping "zero beat" it should not be.

I suppose it was suggested to show this line below to be aligned with the actual frequency. The horizontal line should have an imaginary cross-section with the needle at the point of BFO shift!

73 Bojan

pa7n, 2017-08-03:
Hi Andreas,
In which commit did you fix it?
73 Erwin

db4ple, 2017-08-03:
Hello Erwin,
the problem has not been fixed yet, but analysed. There multiple issues here. One is, that there is a clear bug when using "CW Offset". The frequencies shown below waterfall  are not correct. In addition the frequency bar (horizontal) is not placed correctly unless filter center frequency and sidetone frequency are identical.
And a second one is the fact that the vertical marker line is currently placed in all modes including CW where the  RX 0Hz frequency is. This is "coincidentially" for all modes but CW the TX carrier frequency (even though the carrier is suppressed in SSB).  In CW we receive in SSB mode and the vertical marker shows you the 0 Hz RX but not the CW TX carrier position.

We will fix that, but we will not center the spectrum at CW carrier position. Instead we will move the line to the correct position but keep the spectrum as such center (or not centered as in 1x Magnify). 
Maybe later we implement centering but for some configurations  like magnify 2x and 12khz translate it will not work properly, as there is not enough data to show a full zoomed spectrum depending on the sideband we are using.
Meanwhile I suggest to not use CW Offset, as this provides confusing visualizations (CW TX is better).  
73
Danilo

s53dz, 2017-08-04:
Hi,

Excuse me, please, but now I don't know where you are aiming at.
The only exact frequency that characterizes both SSB modes and CW mode is the carrier frequency. No matter if the carrier is 50 dB down and no matter what demodulation decoding is used in SDR.

There is no side-band in the air when transmitting CW. Only the carrier. So, what is the point of showing any other frequency in a transmission pas-band? In doing this you end up with pointing to a ghost space on a frequency ruler. The BPF center frequency is something else. The operator should know what to do with that. As, for instance, what to do with an IF shift if applied.

73 Bojan

s53dz, 2017-08-04:
... and yes, I think, when hypothetically you are tuned to a (carrier) frequency and there would be a USB and a LSB signal on that same (carrier) frequency at the same time, by switching the modes, you should hear each of them properly and in whole. Without changing the (carrier) frequency.

df8oe, 2017-08-04:
Yes - the only relevant is the carrier.

So we must have
- frequency digits show carrier of RX and TX (already given)
- in scope or waterfall you see vertical "receiving marker line" *always* in center of filter marker
Because of people tend to hear RX at the same tone frequency as they have adjusted their sidetone frequency technically there is an addition/substraction of sidetone frequency and LO. Problem is that if you do not choose a CW filter which matches to your sidetone frequency you will get off center of filter and it is possible, too that you are unable to work using that sidetone frequency because of you are leaving the passband of the given CW filters.

So I think we do not need any modes like TX, RX and SHIFT - only CW-U and CW-L. This is for a smoother switching between SSB and CW without retuning. And maybe we do not need any "receiving marker line". Only the shown passband of the filter. Maybe for better optical identification it would be nice to have TWO receiving marker lines located at the edges of the filter so that you can clearly see if scope peak is inside or at the edge of the filter...

s53dz, 2017-08-06:
Just a report,

in FW 2.5.32 the situation is this:
BFO off = 650 Hz
Dialed frequency = 14.100 MHz
```
displ.mode, display freq., TX freq.
SHIFT, 14.100.000, 14.100.000
TX, 14.100.000, 14.100.650
RX, 14.100.000, 14.100.000
```
The showing of BPF in a horizontal line is ok. But I still don't get it why there is a need to show a vertical needle in CW mode shifted to a ghost frequency offset and not to zero (actual carrier)?
And yes, you have an issue showing frequency at zoom x 1.

73 Bojan

df8oe, 2017-08-06:
What vertical needle shifted to a ghost? I do not see it... Everything shows exactly carrier...

73
Andreas

db4ple, 2017-08-06:
Hello Bojan,
thanks for testing. 

However, I don't understand what you wrote regarding "ghost" offset. The vertical line should appear exactly where the carrier is. It would be great if you could tell me where you measure the carrier and at which frequency point you see the vertical line (best done in magnify x16).  

Another thing: There is no such thing as a BFO in the mcHF signal path. The BFO setting (CW SIdetone Frequency) determines the frequency of the carrier IQ signal being generated. This digitally generated signal is converted to analog in the codec and then mixed with the LO frequency (CW carrier frequency = Sidetone Freq + Si570 LO Frequency / 4)  during transmit.

Technically the sidetone frequency is not relevant at all during RX. What you hear is a filtered SSB signal, I think I mentioned that. The firmware currently always offsets the RX frequency by the CW Sidetone frequency to the carrier (no matter which of the CW offset mode you have). In order to receive where you transmit you have to use a RX filter which has exactly the same center frequency as your CW Sidetone frequency. Only then RX and TX match in CW mode. 

Just for the record: none of the current developers designed the CW part of the firmware. 
I personally (and I think, it is not just me) have trouble with the 3 different modes. 
Now with the RX marker (yes, the vertical line was an RX marker in the original design (also not done by any of the current developers. see the KA7OEI documentation) converted to a TX marker and the source code structured a little  better in this area we could indeed consider a change in the overall approach for CW to something simpler as Andreas suggested.

BTW, the frequency display problem in the bat at magnify 1x has been fixed in 2.5.33   

73
Danilo

s53dz, 2017-08-06:
Andreas, Danilo,

nice work.
I am attaching a picture just taken of a CW display.
You can see the offset being shown in spectrum analyzer window for the RX / TX frequency shown above. I call the vertical line a needle. 

And we should talk here only about this actual frequency. It is only one in CW. And, again, it is no matter how this carrier is produced, either by an oscillator output or by injecting a tone into an SSB channel. HW or SDR.

Please ignore the LCD display artefacts - it is just a plastic cover.

Yes, I am using abrev. BFO on purpose for better understanding the needed offset from an analogue world.

I hope the picture comes thru.
73 Bojan

[image]

db4ple, 2017-08-06:
Hello Bojan,
what I read from your display: TX CW carrier will be at 14100.650 Hz (CW Sidetone is set to 650Hz). RX is at same location since you selected a 300Hz Filter with 650 Hz.  Can you confirm this?
 
If noy and you find your TX carrier at 14.100 you have CW Offset shift mode selected. I haven't fixed the display issue in this mode: the numeric frequency bar display under the scope doesn't show the real frequency scale (it is shifted by the CW sidetone frequency, e.g. 650 Hz in your case). The scope and filter bar should be correctly positioned.

Now for the main frequency display: To achieve correct display of TX frequency in numeric display, you have to select "CW Freq Offset" = RX AND use the filter with 650 Hz center. 

Or wait for the fix for "Shift" then you could also use this mode. I can't recommend "CW Freq Offset" = "TX" this is the most confusing mode (for me at least).

df8oe, 2017-08-06:
Interesting. If I tune to 14.100 and set mode to CW my "needle" is positioned above 14.100 - and not like yours on 14.100.650. I am using "CW Freq. Offset" as "RX" - like Danilo recommends.

Maybe you use another mode or there are some settings (which I never touched) that impacts correct displaying of needle...

73
Andreas

df8oe, 2017-08-06:
Additionally I do not see any argument what other modes than "RX" can be useful for. They are almost confusing and cannot be used for anything...

df8oe, 2017-08-06:
OK - after playing around a little bit I report something that can be done better:

If you are toggling through modes (USB/CW/AM etc.) main frequency display changes if I select "CW" (to something what results in sidetone addition). Main frequency display should not change when toggling modes - sidetone addition must result in a different internal LO frequency (which is not interesting for user).

73
Andreas

db4ple, 2017-08-06:
There is no such thing as free lunch:

Offset -> you can change the CW Freq Offset frequency and the TX carrier frequency remains  unchanged, toggling between modes  (CW <-> not CW) does not change visible TX frequency (but waterfall WILL move)

RX -> you can change the CW Freq Offset frequency and the TX carrier frequency remains  unchanged, toggling between modes (CW <-> not CW) does change visible TX frequency (but waterfall WILL NOT move)

s53dz, 2017-08-06:
I agree, use just RX display mode. RX display mode was used also when the picture was taken.
Danilo, for the actual TX freq. please see my above table.
To repeat: everything works, only the "needle" in CW shows +650 Hz up. TX carrier is at 0 Hz.

73 Bojan

db4ple, 2017-08-06:
@df8oe : My FT-817 does change the displayed frequency when I switch to and from  CW modes.

db4ple, 2017-08-06:
@s53dz: I will test again, to see if  I can verify your results.

s53dz, 2017-08-06:
Yes, I suppose it switches the display, but what is the actual TX frequency than? It switches that too?
I have measured the RF output frequency with a counter in each case.

db4ple, 2017-08-06:
My test result :
CW TX is at 29.000.000 Mhz ((using a FT-817 w/ TCXO)
CW RX is at 29.000.000 Mhz. Tested using a GPS controlled "frequency generator" (Ultimate3S)

[image]

[image]

df8oe, 2017-08-06:
I can confirm Danilos test results. If have checked with counter, too.

s53dz, 2017-08-06:
I can confirm all of that either. Except one thing, how come that your number band under waterfall is aligned to 0 (ok, to 50 Hz) and mine under a spectrum is 650 Hz off (my offset is 650 instead of your 750)? Is it later than  FW 2.5.32?

db4ple, 2017-08-06:
Hello Bojan,
I was using 2.5.33. But now I fixed the frequency display for CW Freq Offset == Shift.
wait for Andreas to merge my last pull request ( it will be in 2.5.34).
Here CW Freq Offset == Shift and CW Freq Offset == RX work as designed, each has slightly different behavior when it comes to the waterfall or scope display, see above. But both now produce the results I expect. Displayed frequency is always telling you the TX carrier frequency, horizontal filter bar is where it should be in the spectrum. TX Frequency marker is where it should be. Frequency bar is also correct.

For me, this part is finished, now you and other should validate the result.

s53dz, 2017-08-06:
Danilo,

Yes, I confirm in FW 2.5.34 it is the same as you pictured there. For me this is all aligned now. So, issue gone.

TNX 73 Bojan

db4ple, 2017-08-06:
Hello Bojan,
thanks for confirming. 

Just for the record: While we fixed several longstanding issues on the go, the actual cause for all of this was a change of existing functionality (RX Marker -> TX Marker). I however agree, that the change makes sense (otherwise I wouldn't have spent my time on it). 

I am wondering, what you (and also Erwin, the original reported of the problem) think about two things:

Shall we remove the "TX" CW Freq. Offset Mode? It seems to be useless as the displayed numeric frequency does not match the cw tx carrier frequency.

And what do you think: Should we couple rx filter center frequency to CW sidetone frequency? The way it would work -> you select a filter, and the center frequency of that filter is automatically applied as sidetone frequency (i.e. as CW TX frequency).  I am not yet sure if this is easy to do or even if it makes sense at all to you CW guys.  
@pa7n : Could you please confirm that you are satisfied with the implemented solution?
TNX,
Danilo

s53dz, 2017-08-06:
Danilo,

I have just had time to go to check again. Here is a new/old table showing the displayed and actual RX and TX frequencies.

columns = display mode
rows = displayed, RX and TX frequency recpectively

```
         SHIFT,  RXmode,  TX mode
FREQ.
--------------------------------------
DISPLAY, 14.100, 14.100, 14.100
RX FREQ, 14.100, 14.100, 14.100
TX FREQ, 14.100, 14.100, 14.100.650
--------------------------------------
HOR.BAR, shifted ok, shifted ok, centered ok!

```
Considering this CW result, I am happy with the RX display mode as is. RX = TX = DISPLAY. As a matter of fact I would prefer to see everything at the center, hi. 

For your second point, I see some relevance there but I think the operator should know what he/she is doing when choosing filters.

TNX 73 Bojan

df8oe, 2017-08-06:
In magnify x 1 we NEVER have enough data to show needle in center,
in magnify x 2 we do have enough data IN SOME CASES to show neelde in center,
in magnify > 2 there is always enough data to show needle at center, but:

IT IS MUCH MORE COMPLICATED to code that. Issue is gone, I do not hink it does make sense to invest much power to do just this optical changing ;)

vy 73
Andreas

s53dz, 2017-08-06:
Andreas, I meant it only as a beauty thing, although it would be nice to have it in the middle. But ok.

I have spotted that the old "needle" vertical line is not cleared when on new position. The same now happens when using RIT.

db4ple, 2017-08-06:
Hello Bojan,
do see this in Waterfall mode (probably not)? In Scope mode there are still some issues, since the old code was very much centered around using the grid lines only for the carrier line.  If reporting these issue, it is helpful to get a little background such is scope or waterfall being used, which magnify shows the problem etc. 

Workaround: go into menu and back. 
We will fix that, too. But this is a fiddly thing.  Waterfall was much easier in this respect.

s53dz, 2017-08-06:
It is in scope of course. In waterfall you calculate it dynamically so it is gone with the pace of the fall.
It occurs in every zoom of scope.

I can understand it is quite a "music" when cleaning this.

s53dz, 2017-08-07:
When exploring the behavior of the last FW I went to check the initial meaning and implementation which was well described in Clint's Operating Manual. I must say he describes the old and new way of displaying frequency (shift) quite well. Although I still do not see much worth in displaying the frequency which is not TX. Displaying RIT, XIT and SPLIT is something else! 

However, this part draw my attention:
> Pressing-and-holding button G3 when in CW, LSB or USB mode will generate a tone that is
> equal in frequency to the CW sidetone and transmit-receive offset. This may be used to “spot”
> the frequency so that you can transmit on the same frequency as the station with which you are
> communicating. The loudness of this tone may be adjusted using the “Adjustment Menu”
> option “Beep Volume”.

I must admit I have completely overseen the option to SPOT ON the signal by holding the G3!!!
I suppose I have to check this behavior in the light of the latest changes.

db4ple, 2017-08-07:
Hello Bojan,
I have bad news for you: Press and Hold G3 will do nothing. Currently there is old code there refering to AM_experiment (not for public use it says) AND even the code for the AM_experiment is long gone.
There is a comment mentioning the reference tone generation but not code.  In the 18+ month I have been working with the firmware, there never was anything like the mentioned reference tone generation in this place. Sorry. You have to use the visual path (using the tx carrier line).

s53dz, 2017-08-07:
Danilo,

Thank you for this. OK then. No worries about that.

By yesterday's testing the new frequency display in scope and in numbers I have noticed that in CW RX display mode the frequency band under scope is being shifted according to RIT setting. In this light I would then go for a CW TX display mode, since it is TX-centric. As this is the nature of RIT, going around the TX frequency. If we would have XIT, then the RX-centric mode would be prefferable.
Please see the attached drawing to clarify.

[image]

But here again, perhaps the display evolution is good (at least for me) but then, we are yet further from a simple remark of Erwin:

> Another possibility is to display the real transmit frequency exactly in the middle of the scope. As it is now the transmit frequency seems like it's in the middle of the screen but it's offset with the CW tone frequency.

df8oe, 2017-08-07:
Just brainstorming: we can show one constant line showing TX and RX (if RIT == 0 and no split operation). If there is RIT != 0 OR split operation there can be a second, dashed line that shows RX.

s53dz, 2017-08-07:
And if I am already at drawing, here is a "wish display", where a TX frequency is also shown in numbers and the bottom-line marks the filtered receiver window:

[image]

s53dz, 2017-08-07:
I can put some observations when testing FW 2.5.37.

Excellent job showing frequency and the RX line beneath. And moving the filter (RX) line to the center.

However, when I turn RIT in + direction a green needle moves to the left, and numbers on the ruler increases at the same time. Whereas the horizontal filter line stays.
And vice versa, when RIT turns to -, needle moves to the right and numbers decrease.
I think the ruler numbers (and the scope) should stay, only the RX needle and filter line should move with RIT. To the right when RIT + and to the left when RIT -.

I think the CW needle should be at zero point as it is in USB/LSB mode. I don't care even if there is some small grey area (no FFT data) at the side of scope because of that.

s53dz, 2017-08-07:
I am sorry, my mistake: The needle (green in my case) is TX frequency and not RX at all. I looked again in menu. So, this TX needle should be at a fixed position then. RX window position should move by switching modes and/or by using RIT and not TX frequency.

I suppose by setting CW TX display mode, I should have such TX frequency scenario shown.

Some remarks:
1. There are two fields for showing band. Probably only one or none needed. Is it there for cross-band?
2. There is a big number label showing TX frequency all the time (which is ok)
3. There is small RX frequency shown above that all the time (which is ok and working nice)

4. This picture shows TX needle offset in CW mode.

[image]

TNX GN Bojan

db4ple, 2017-08-07:
As discussed, the display is centered around the receiving spectrum which is based on the LO frequency, and RIT moves the receiving frequency thus the LO thus the spectrum display. There is a fixed relationship between RX frequency and LO frequency in the UHSDR firmware.
We already discussed, that currently I (or Andreas) will not invest in a centered TX frequency display. You may want to open another issue for that so that it does not gets lost, but it will not be done unless someone spents some time on it (read significant amounts). 

For the band question: 
https://github.com/df8oe/UHSDR/wiki/Firmware-FAQ#q-why-is-the-band-shown-twice-in-the-ui

BTW, I think placing an RX carrier marker in the frequency bar could be doable. We could also do 2 vertical lines but not sure if this will improve things and it would be also a lot of work (not as much as the centered TX, though).

s53dz, 2017-08-08:
Danilo,

Thanks a lot for this explanation. I mistakenly thought that to align this it was just a matter of introducing a proper index to the x-offset. So I will stay with the CW RX Display mode for the time being.

Concerning the two times Band marking: I rose that Q because of the behavior when using CAT. In my case by changing the frequency over CAT the band mark doesn't change.

BTW: I like scope zooming, filter-line showing the RX window vy much. I suppose this is trend-setting.

TNX 73 Bojan

df8oe, 2017-08-08:
I think I will slightly change that behaviour. If you do *NOT* have set "CAT in sandbox" there must be correct band indicator (like Bnd80m). But if you have set "CAT in sandbox" *AND* you have done a frequency change via CAT the indicator should no longer show band indicator like "Bndxxm" but the word CAT. This will indicate that the frequency you dialed via CAT is NOT applied to the band storage place where you have started from. I will add this today - it will be included in the next daily snapshot.

73
Andreas

db4ple, 2017-08-08:
Hello Andreas,
yes, I agree to that general idea. I have been thinking of implementing a "CAT" VFO memory, which is automatically selected it CAT in a sandbox is used AND a frequency change is made via CAT.
This CAT memory lives on top of the currently selected memory, i.e. band- or band+ will bring you back to the previous band vfo memory and the next will bring you as usual to the lower or upper band.
We could also implement CAT Band Autoselect -> the CAT frequency change goes to the most appropriate band memory (Alternate mode, not replacement of sandbox).

df8oe, 2017-08-08:
Hi Danilo,
indicator as described is published in latest daily snapshot. Nice idea to change menu entry to "CAT Mode" and offer the choices
- none
- in sandbox
- jump to band

There is no wish existing to have such a possibility and I am not sure if anyone wants to have it. So I want to stay at the actual existing structure and wait if someone wants to have it... I am satisfied with the already given CAT possibilities and do use "CAT in sandbox" all the time.

df8oe, 2017-08-08:
Additionally I do not see any sense for a "CAT storage place". When should it be recalled? If any CAT-command happens? Then you configurate your recent settings e.g. in fldigi with some older ones you formerly have selected via CAT...

s53dz, 2017-08-08:
Reporting the FW 2.5.40:

1. You have inverted the FFT spectrum.
2. For me now the only reasonable display mode is SHIFT.

Ad 2:
The numbers stay, the spectrum is aligned.
All other modes are shifting the displayed frequency which is annoying and to me un-understandable.

db4ple, 2017-08-08:
Ups. You are right, the scope is now oriented the wrong way, waterfall is correct.

s53dz, 2017-08-08:
I suppose it is not oriented the wrong way but rather inverted, meaning the center is split to at both sides.

db4ple, 2017-08-08:
Each half is swapped in its orientation, to be precise.

s53dz, 2017-08-08:
Yes,
I think this is raw data output as FFT itself functions.

df8oe, 2017-08-24:
I think this is fixed now. If you think so, too please close this issue.

s53dz, 2017-08-26:
As I said, for me using the SHIFT display option this issue is gone, although with all the remarks I have put. But this issue was started by Erwin.

Now, when the second marker line is so well introduced in RTTY mode, why not extend it to an RX dial marker beside the TX one? I think this may help to a clear view when using RIT. I don't know for SPLIT.

73 Bojan

[3 more replies]

closed 2017-09-13 by pa7n