df8oe/UHSDR #930: S-meter not working with small bandwidth filter in CW

pa7n, 2017-07-02:
Hello,
Another CW thing! :-( 
I made a video about this bug.
Sorry for the moire effect. As I like the screen of the mcHF very much the camera doesn't :-(

[youtube video about s-meter bug](https://youtu.be/q8d98h1L3uc)

73, Erwin

db4ple, 2017-07-02:
Hi,
this bug is known, and has been discussed in the german group. Unfortunately, no one has been making a github issue.

See https://www.amateurfunk-sulingen.de/forum/index.php?board=15;action=display;threadid=754
(Yes, it is in german, I know). Basically in CW (and SSB) the filtering for S-Meter calculation is not always configured correctly, and because of that, the wrong value is displayed.

pa7n, 2017-07-02:
Oke, I didn't read that forum topic. Is someone working on a fix?
73, Erwin

DD4WH, 2017-07-02:
Hi Erwin,
Markus, DL8MBY, spotted the bug and was preparing to fix it. However, at the moment it seems he is busy with another bug. If you want to try a quick fix, use something like this (untested!):

uint16_t offset = width / 2;

if(FilterPathInfo[ts.filter_path].offset) // look into audio_filter.h for all the variables in the FilterPathInfo array
{ // fetch centre frequency of the filter from the FilterPathInfo array, if centre frequency is different from width/2
      offset = (float32_t)FilterPathInfo[ts.filter_path].offset;
}

für SAM_L, LSB, CW_L
Ubin = (float32_t)posbin - roundf((offset - width/2)/bin_BW);
Lbin = (float32_t)posbin - roundf((offset + width/2)/bin_BW);

für SAM_U, USB, CW_U
Ubin = (float32_t)posbin + roundf((offset + width/2)/bin_BW);
Lbin = (float32_t)posbin + roundf((offset - width/2)/bin_BW);

für SAM/AM
Ubin = (float32_t)posbin - roundf(width/bin_BW);
Lbin = (float32_t)posbin + roundf(width/bin_BW);

in the file ui_spectrum.c, alter the function UiSpectrum_CalculateDBm

Have fun!

73 de DD4WH

db4ple, 2017-07-02:
hi,
i will commit a hopefully proper fix soon.

pa7n, 2017-07-02:
It's working! Great and fast fix!

db4ple, 2017-07-02:
Hi Erwin,
since you are very particular about CW, you should now try the new CW Weight Function. I don't know enough about CW to test it or even understand why CW weighting it is a useful thing, but maybe you?
In the CW Mode menu you can now change the weight (not saved because I'll implement the permanent storage only if the CW weight is working and liked by the CW folks).

73
Danilo

pa7n, 2017-07-02:
Hi Danilo,
I'm not a very CW expert. But it has something to do with high speed CW. That could be better readable when changing the weight. But there is some controversy about what CW weighting is. Some say it's the dot-dash ratio normally 1:3 but could be changed from 1:2 to 1:4.5.
Some say it is the length of the dots and dashes in relation with the space in between. If you increase the length of a dot and a dash your average power goes up. That might be an advantage in poor conditions. 
73, Erwin

db4ple, 2017-07-06:
Hi Erwin,
can we close this? Please do so then. Thank you!

pa7n, 2017-07-06:
Yes we can!

closed 2017-07-06 by pa7n