df8oe/UHSDR #932: Variable CW Weight

db4ple, 2017-07-02:
Ratio between dit/dah and pause should be configurable. It is configurable in a number of TRX, so why not here?
Has been requested in Yahoo NG. With the now fixed timing, we have good base for supporting proper weighting.

pa7n, 2017-07-02:
Nice this feature is coming to the mcHF :)
But I think the current version of this implementation is not working. Strange things happening when set on a other value then 100.  The keyer continuously kept transmitting.

db4ple, 2017-07-02:
Hm. Will look into this.

s53dz, 2017-07-04:
Hi,

There are many opinions among hams of needing and not-needing changing the weighting but I think it has to be done right. I have found a good explanation of doing the "proper" weighting of CW signal. It is from PA0WV on some forum. Maybe it helps here:

> When you define a dash as 3 dits mark and 1 dit space, and you define a dot as one dit mark and one dit space , then the dashmark dotmark ratio is 3.
> 
> However you can add  a percentage of the dit time to the dot mark and substract the same time from the dot space time.
> So you can also do with the dash. That is defined as weighting. Because dot and dash time (including their accompanying space) don't change, this operation does not affect the Morse speed relation with the unmodified dit time..
> 
> The total dash time doesn't change in that case, because what was added to the dash mark is subtracted from the accompanying dash space.
> 
> So weighting is a dot related percentage, not dit related, in order to keep speed constant with dash marks of 3 dits, it is impossible to go below 2 dits for a dash-mark en 2 dits for a dash-space.
> 
> Another thing is changing the dot dash ratio,  when you make it larger dashes grow and dots are smaller (remember: including their accompanying spaces) That will make your speed always character dependent. (think of figures 0 and 5) I can't see a reason for doing that when weighting is already adjustable.
> 
> When it is included in a design, I suppose it is done just because "You asked for it" .

73 Bojan S53DZ

db4ple, 2017-07-04:
Hi Bojan,
thanks for the text above. Implementation in the firmware is exactly like the text says. 

Based on the select WPM we calculate wpm_dit,wpm_dash and wpm_pause time according to 1:1 and 1:3 ratio. Then we use the weight value to calculate an correction value (corr = dir * (weight -1.0)), which we subtract from wpm_pause and add to wpm_dit and wpm_dash -> e.g. real_pause = wpm_pause - corr, real_dit =wpm_dit + pause. So the WPM timing does not change since: real_dit + real_pause = wpm_dit + corr + wpm_pause - corr = wpm_dit + wpm_pause.

db4ple, 2017-07-05:
With the permanent storage of the weight we close this issue.

PR #936 (2017-07-05): Make CW KEYER WEIGHT setting permanent (and other CW refactorings)
besides the persistent storage, some names of defines and variables for cw
have been aligned.
This closes #932
closed 2017-07-05 by db4ple